prusa3d/PrusaSlicer #921: Configuration Wizard not executable 1.40.0-beta

DanyboyInSwitzerland, 2018-05-24:
### Version
1.40.0-beta+win64

### Operating system type + version
Win10Pro, Version 1709, Sys-build 16299.431

### Behavior
Updated from here:
https://github.com/prusa3d/Slic3r/releases/tag/version_1.40.0-beta
Copy-paste into existing folder program:
C:\Program Files\Prusa3D\Slic3rPE
Upon first execution of SlicerPE, a pop-up asks me to execute the configuration wizard, offering the option to directly start the wizard. Upon starting the wizard, application crashes to desktop without any further messages.
When I execute SlicerPE after that, it runs fine, version-number is correct and also the different profiles are available for selection, operates fine.
If I start the configuration-wizard, I get the following error:
An unhandled exception occurred. Press "Abort" to terminate the program, "Retry" to exit the program normally and "Ignore" to try to continue.
Abort recreates the same message, after clicking abort a second time Slicer does no more respond and gets closed by windows.
Retry closes and exits Slicer to desktop without any further messages.
Ignore lets Slicer continue as if nothing hapened.

#### STL/Config (.ZIP) where problem occurs
No stl was selected at that point, Slicer was just started.

Slicer Config Directory attached as per request by Vojtech Bubnik in Facebook:
[slicer_config_directory.zip](https://github.com/prusa3d/Slic3r/files/2035911/slicer_config_directory.zip)

DanyboyInSwitzerland, 2018-05-24:
> Don't you have incidentally some non-Latin1 characters in your user name?
> What is the full path to the Slic3rPE folder? Do you use any redirect
> points inside the path? Isn't your home directory on a network drive?
>
> What happens if you start slic3r from the command line with the
> --datadir=c:\Slic3rPE path? It should run the configuration wizard and
> proceed with the installation successfully. What happens if you close
> Slic3r, delete the Slic3rPE directory (the one in your user/Roaming/AppData
> directory) and start Slic3r again? Will the wizard crash?
>
> Thanks for your help
>
> On Thu, May 24, 2018 at 5:08 PM, DanyboyInSwitzerland <

Hello Bubnikv,

Q: Quite some questions!
A: But I'm glad to answer if it helps...! :)

 

Q: Don't you have incidentally some non-Latin1 characters in your user name?

A: Negative: User is Daniel.

Q: What is the full path to the Slic3rPE folder?

A: C:\Program Files\Prusa3D\Slic3rPE

 

Q: Do you use any redirect points inside the path?

A: Not as far as I'm aware - standard-installation with the Prusa-Package. Since I could not find any executable in terms of setup in the package, I updated by copy-paste the files of the new version into the Slicer-folder. Maybe that created the problem in the first place?

 

Q: Isn't your home directory on a network drive?

A: Negative? All is local. From your question I interprete you find this unusual? Shouldn't this be so?

Q: What happens if you start slic3r from the command line with the
--datadir=c:\Slic3rPE path? It should run the configuration wizard and
proceed with the installation successfully.

A: I am not sure whether I can follow your instructions, so let me tell you what I did...
From the commanline, I executed the following:

C:\Program Files\Prusa3D\Slic3rPE>slic3r.exe --datadir=C:\Slic3rPE

There was a short wait of a few seconds, then a window with slicer popped up and closed itself almost instantenously without any interaction, after that nothing happens.

 

Q: What happens if you close Slic3r, delete the Slic3rPE directory (the one in your user/Roaming/AppData
directory) and start Slic3r again? Will the wizard crash?

A: I am sorry, but I might be lost - I cannot locate that AppData-directory you mention.

Or maybe it's non-existant and that's precisely the problem?
However, the only thing in any user-directory is the folders for the slicer settings, under C:\Users\Public\Documents\Prusa3D

 

I hope this helps nevertheless...

 

Cheers,

Daniel

 

Gesendet: Donnerstag, 24. Mai 2018 um 18:03 Uhr

bubnikv, 2018-05-24:
> Since I could not find any executable in terms of setup in the package, I updated by copy-paste the files of the new version into the Slicer-folder. Maybe that created the problem in the first place?

That may be the issue.

What happens if you extract the binary zip file on your desktop and run the slic3r from there?

Primermecos, 2018-05-25:
I've run into the same problem and I assume, like bubnikv wrote, the copy&paste into the same folder causes this. Once I put the files in a new folder, I was able to run the wizard.

vojtechkral, 2018-05-25:
@Primermecos Can you tell me what Slic3r release was the one that you copied the new one over? Perhaps we may be able to reproduce the problem that way and cosider if it could be fixed ...

Primermecos, 2018-05-25:
I think I extracted "Slic3rPE-1.40.0-alpha1+win64-full-201805141300"in the default Prusa3d-folder. (C:\Program Files\Prusa3D)

vojtechkral, 2018-05-25:
@Primermecos yes, but do you remember which was the previously installed version?

Primermecos, 2018-05-28:
It was the last compiled released version. Think it was Slic3rPE-1.39.2.win64-full-201805041535.zip. I may be wrong, but I think that's not important, since both versions were installed in separate folders and can run at the same time. Anyway, I hope I could help.

vojtechkral, 2018-05-28:
@Primermecos Thanks!

vojtechkral, 2018-06-04:
This should be fixed with df70ee14322959740298ddf7b676614fdf431cdf , if problems persist, please let me know.

closed 2018-06-04 by vojtechkral