prusa3d/PrusaSlicer #922: [Request] Estimated print time / filament usage in output name

danielegger1, 2018-05-24:
### Version
1.40.0 Beta

It would be nice to have somewhere a checkbox to activate additional output names like estimated time and required filament.
Currently I am doing this by hand, but when I use "Send to printer" it is quite awful to add a additional name after the export (but this is the time where I get the estimated values).

What I would like to have is the original name in the output + estimated time in a defined format:
Import Filename = Ankly
Output Filename = Ankly16H12M

Same with filament
Import Filename = Ankly
Output Filename = Ankly20m

Combined it could look like
Output Filename = Ankly16H12M20m

This would be a quite nice feature as I am prepairing the next 10-20 files to print but I need to change the order of the prints sometimes as it depends on the times when I am back home to change the print. But for this I need to know the roughly estimated times / filament usage.

Chrismettal, 2018-05-25:
I second this. Should be a matter of saving the time as a placeholder for the output file format like the parameters. Or is there already a placeholder that i just don't know of?

WhiteDragonD, 2018-06-22:
I'd like to 3rd this.  Every time I save a file I have to manually add the h/m to the name.

Name(0.15mm)-3h40m-2Brim_0Support_cubic.gcode

`[input_filename_base]([layer_height]mm)-hm-[brim_width]Brim_[support_material]Support_[fill_pattern].gcode`

I looked at the site: http://mauk.cc/mediawiki/index.php/Slic3r_placeholders but didn't see a placeholder for time estimation.

macdarren, 2018-06-22:
Maybe related:  I would like to see the standard slicer function where I can simply get a dialog box to adjust the G-Code name when I send things to my printer (octoprint) from the GUI.  Just so a whole plate full of parts isn't named for the first one....maybe I am missing something but the non-Prusa version prompts me while the Prusa versions do not.

tracernz, 2018-06-22:
@macdarren That is done in latest beta (1.40.1). ;)

gsedej, 2018-10-23:
May I ask why there is no "sliced info" when hitting "Slice now"?
 The "sliced info" only appears when outputting the file. (e.g. estimated time printing, filament cost, filament used...)
Using 1.41.1-rc

bubnikv, 2018-12-12:
The Slic3r PE 1.42 alpha, which will be released soon, supports now the following additional placeholders in the output file template:

        "print_time", "normal_print_time", "silent_print_time", 
        "used_filament", "extruded_volume", "total_cost", "total_weight", 
        "total_wipe_tower_cost", "total_wipe_tower_filament"

The time is in the dhm format (5d3h1m for 5 days, 3 hours and 1 minute), rounded to a full minute.
Closing.

closed 2018-12-12 by bubnikv