prusa3d/PrusaSlicer #923: Octoprint Upload IP is not saved between sessions

pricedev, 2018-05-25:
### Version
Slic3r Prusa Edition
Version: 1.40.0-alpha1+win64
Build:   Slic3rPE-1.40.0-alpha1+win64-full-201805141300
Operating System:    MSWin32
System Architecture: MSWin32-x64-multi-thread

### Operating system type + version
Microsoft Windows [Version 10.0.17134.48]

### Behavior
On the Printer Settings tab, for a custom printer profile, the OctoPrint Upload / Hostname, IP, or URL field is not saved when an IP is entered and the profile is saved, followed by closing and reopening Slic3r.   

After reopening Slic3r and returning to the profile, the field is blank.

_Is this a new feature request?_ No

#### STL/Config (.ZIP) where problem occurs
NA

tracernz, 2018-05-25:
This one seems to get a new issue each day! Fixed in https://github.com/prusa3d/Slic3r/commit/35bf4aee7df3baee5f64b769d58d20199bb5268a. Try the latest release https://github.com/prusa3d/Slic3r/releases

vojtechkral, 2018-05-25:
Yup, #869

vojtechkral, 2018-05-25:
@tracernz On the bright side, it's a good indication of how popular the feature is ...

closed 2018-05-25 by vojtechkral

pricedev, 2018-05-25:
Works great.  Thanks!