prusa3d/PrusaSlicer #924: Interface reorganization (quick buttons)

BastR, 2018-05-25:
### Version
1.40.0-beta+ 

### Operating system type + version
macOS High Sierra 10.13.5

### Behavior
Hello, is it possible to display "export stl, etc." buttons vertically instead of horizontally or in the quick access bar? The loss of space for viewing the bed is very important, it is even more so with the translation of the interface. You can see on the screenshots that a lot of space is lost unnecessarily and on the contrary on quick access bar space is unused. Some buttons could also be removed from the quick access, I think in particular of "export stl" which is present when a right click and in the top menu bar too.

[image]

YuSanka, 2018-05-27:
Thanks for your report. We are working on upgrading/reorganizing of the main UI page now. We plan to implement most of the user's suggestions in the next version.

BastR, 2018-09-02:
A small update, after reading the documentation I reformulated the French translation of the buttons so that they are smaller so there is more space for the bed preview. I took the opportunity to translate all the sentences introduced by the new version of Slic3r PE. 

This is the first time I've done this, if I understood correctly I have to propose a PR of my fork?

**Before:**
[image]

**After:**
[image]

bubnikv, 2019-01-29:
Please check the latest alpha, the layout changed significantly.
https://github.com/prusa3d/slic3r/releases
Closing.

closed 2019-01-29 by bubnikv