prusa3d/PrusaSlicer #925: Problem when minimizing Slic3r on Mac OS

kjetilengstrm92, 2018-05-26:
### Version
1.40.0-beta+

### Operating system type + version
Mac OS 10.13.4
### Behavior
* _Describe the problem_
When minimising application clicking the icon in the dock does not open it again.

I have to right click the icon and "show all windows" to open it again.
  * _Screenshots from __*Slic3r*__ preview are preferred_

#### STL/Config (.ZIP) where problem occurs

[image]
[image]

mvasilakis, 2018-05-26:
To confirm I too have this issue. I've been meaning to open an ticket for it as well. The windows should be restored when clicking on the dock icon as OP stated.

stelgenhof, 2018-05-27:
Interesting issue. I am running MacOS 10.13.4, Slic3rPE 1.40.0-beta as well and I can minimize Slic3r (using the orange circle icon of the main window) and reopen Slic3r when press the icon in the Dock Bar.

kjetilengstrm92, 2018-05-27:
I have to add that this is not new to Slic3r 1.40.0. It was also present in 1.39.

stelgenhof, 2018-05-27:
@kjetilengstrm92 I also tried it with 1.39.2 on my Mac and have no issues. Perhaps I am not doing the same thing you are doing? :)

stelgenhof, 2018-05-28:
@mvasilakis After seeing your video, I believe the issue is just with the minimize (Orange) button. I've tested this with the default Apple Maps Application:

1. Selecting "Hide" or Command+H: The app is hidden and can be made re-appear (Maximized) by clicking on the Maps icon in the Dock
2. Selecting "Minimize" (Window Menu), Command+M or clicking the orange minimize button: The app is minimized and appears as an additional icon in the Dock. Clicking either the Maps icon or the minimized application, makes the app re-appear (Maximum).

With Slic3rPE:
1. Selecting "Hide" or Command+H: The app is hidden and can be made re-appear (Maximized) by clicking on the Slic3r icon (Black/Orange) in the Dock
2. Selecting "Minimize" (Window Menu), Command+M or clicking the orange minimize button: The app is minimized and appears as an additional icon in the Dock. Clicking the minimized application, makes the app re-appear (maximum). However, clicking the Slic3r icon (Black/Orange) in the Dock, doesn't make the app re-appear (maximized) again.

I have tested this with both v1.39.1 and v1.40.0-beta, and neither work (app gets maximized again) when clicking the Slic3rPE icon.

bubnikv, 2018-06-04:
Isn't this issue a duplicate of https://github.com/prusa3d/Slic3r/issues/129 ?

stelgenhof, 2018-06-04:
From what I can tell, it looks like a duplicate of #129

bubnikv, 2018-06-07:
The problem is most likely deep inside the wxWidgets OSX binding,  under
src/osx/cocoa/utils.mm

- (BOOL)applicationShouldHandleReopen:(NSApplication *)sender hasVisibleWindows:(BOOL)flag
{
    wxUnusedVar(flag);
    wxUnusedVar(sender);
    if ( wxTheApp->OSXInitWasCalled() )
        wxTheApp->MacReopenApp();
    // else: It's possible that this function was called as the first thing.
    //       This can happen when OS X restores running apps when starting a new
    //       user session. Apps that were hidden (dock only) when the previous
    //       session terminated are only restored in a limited, resources-saving
    //       way. When the user clicks the icon, applicationShouldHandleReopen:
    //       is called, but we didn't call OnInit() yet. In this case, we
    //       shouldn't call MacReopenApp(), but should proceed with normal
    //       initialization.
    return NO;
}

Until we have a chance to debug it reasonably, it is quite difficult to solve this issue. I personally have wasted about two or three working days with this issue with no resolution. We will have more chance to fix the issue once we get rid of the Perl bindings, so we would have a chance to debug the wxWidgets.

wabbitguy, 2018-06-08:
I use 1.3.9 thru 1.4.0B on a Mac Mini, 10.13.5, using the minimize orange button. Window goes to the doc, clicking on the icon on the dock and the window pops back open. No issue at all.

My dock preferences setting are right side, and "Minimize windows into application icon" is unchecked. Thus windows minimized to the area at the end/bottom of the dock.

stelgenhof, 2018-06-09:
@wabbitguy The issue described here is when you click on the Slic3r logo after you've minimized the application. Slic3r doesn't come back up again. Other applications like for example Safari comes back up again when clicking either the Safari logo or the minimized Safari instance in the Dock.

[image]

In the above screenshot: clicking the minimized Slic3r instance (green box) works, while the Slic3r app (red box) doesn't.

wabbitguy, 2018-06-09:
@stelgenhof Thanks for the clarification on that... I've never done it that way, I always use the "green boxed" one in your snapshot because it's a smaller selection "area" so I find what I want faster. Creature of habit...