prusa3d/PrusaSlicer #926: Support Material bug

danielegger1, 2018-05-27:
Difference in Support Material generation - it works fine in the older version
Version 1.39.2 
Version 1.40.0 Alpha

Example with the issue:
https://www.thingiverse.com/thing:39751

Filename:
OpenRC_Truggy_Rim.A.1.stl

Settings:
Generate Support Material + Contact Distance Z = 0.1 (also the default 0.15 does not work correctly)

Attached are the same files sliced in the two different versions.
In the older version the distance between the support material and the part itself on a certain place is correct, but in the new slic3r there is a distance of 0.5mm instead of 0.1 which ruined my print in that case.

[image]
[image]

FidelCapo, 2018-05-28:
Unfortunately, I am not able to reproduce the issue you described. Can you please provide AMF or 3MF files where the issue is occurs? Thank you

bubnikv, 2018-05-30:
Would you please install the beta and export the AMF file through the beta? There is a bug in alpha, which breaks the exported AMF file if there are XML delimiting characters stored inside the custom G-code. The beta correctly escapes these characters. Thanks.

bubnikv, 2018-05-30:
Namely I don't understand the second screenshot. I would be interested into the data to reproduce it.

dagobertch, 2018-05-30:
[image]
[image]
[image]

I have the same problem. Last support layer at 9.6, first model layer at 10.1 with contact Z value 0.1. 
this was the model: https://www.thingiverse.com/thing:2782540

I tried with and without interface layers, no difference. I had to turn on the „support bridges“ option, because the slic3r wanted to make very long bridges

dagobertch, 2018-05-30:
But my problem was in 1.39.1 and 1.39.2. have not tried the 1.40.0

bubnikv, 2018-06-04:
> I have the same problem. Last support layer at 9.6, first model layer at 10.1 with contact Z value 0.1.
this was the model: https://www.thingiverse.com/thing:2782540

That's because the bottom most object layer is printed with a bridging flow by Slic3r as it is being printed into a thin air. Therefore the first model layer is 0.4mm (the nozzle diameter) + 0.5mm (the Z contact gap) above the support top surface. This is intended. You may get more understanding about the current behavior and limitations of the supports from https://github.com/prusa3d/Slic3r/issues/102

Massa0815, 2018-06-05:
I think its the same Problem as in issue #885.... I never got the support z distance below 0.3 mm. Small parts, especialy round parts, alway fall of the support. They wobble more and more with the ammount of layers, because the aren't supported by the support. This problem ist maxed at sloped parts.

LubosD, 2018-06-07:
I've been analyzing why supports work so bad in Slic3r for me (unlike in Cura) during my ABS prints and I've discovered this along with an identical issue from 2015 [reported for the original Slic3r](https://github.com/slic3r/Slic3r/issues/2930).

I know the devs throw calculations at everybody, but right now I can either have a 0 gap between the supports and the object or an 0.6+mm gap (0.4mm nozzle+contact Z distance rounded up to layer height), which is about as crap as having no supports at all, because the perimeters and all get printed in thin air. What I need is a 1 layer gap.

Seriously guys, this needs to change. Please, pretty please, make it stop being smart, make it stop adding the nozzle diameter to the _Contact Z distance_.

Massa0815, 2018-06-11:
I think if you dont bring the Support of Slic3r to work better, all your new features are not worthy. I Love Slic3r but the support gap is a big problem.

BrandonMaynard545, 2018-06-29:
having the same issue

thom-w16, 2018-07-02:
Having the same problem. It looks like in the original Slic3r you can modify some files to fix this but that is not the case with Prusa Slic3r.

Massa0815, 2018-07-06:
OK .... how i understand the problem: Bottom of part above the support uses the Bridge flow ratio. This is calculated by nozzle diameter. If i use a 0.4 mm nozzle. i get at least 0,4mm distance between part and support.
Why can't you create an option (checkbox) to disable the bridge-flow ratio for bottom of the part above support material. And then you can use the user setted distance. 
The Main Problem is, that printing mechanical parts with big areas above support works well. But if you want to print organical shapes with small round parts above the support the parts wont hold at the support and allway fall of the support. i cant print for exampe Mini figures for tabletop games with slic3r because of this absolute BIG BIG  problem.

Massa0815, 2018-07-11:
Look at this gaps. This Support is worthless
[image]

vojtechkral, 2018-07-30:
I think this is the same problem as #102 , right?

In any case, this is being looked at, however the support generation needs an overhaul anyway and it's taking some time...

ubershy, 2020-11-18:
A kindly reminder that this behaviour is still unchanged and does not satisfy many people.
Many words have been spoken about it. My own conclusion - it's not a feature. It's a bug. It doesn't matter if calculations are correct if they are applied based on incorrect premise.

bubnikv, 2021-03-18:
To improve quality of the object over support, we have added a new settings: "thick bridges". If enabled (that is by default), PrusaSlicer behaves as before. If disabled, bridges are printed as in any other slicer: The extrusion rate is given by the normal layer height x extrusion width * bridge flow ratio. 

[image]

Implemented with ceea9de8b8b1d4b35bdd1180051033680c405ba1 f01f02154c617cc24835475da419ed6322f2ceec

closed 2021-03-18 by bubnikv