prusa3d/PrusaSlicer #927: Skirt intersects multi-body object

zbrozek, 2018-05-28:
### Version
1.40.0-beta

### Operating system type + version
Windows 10 April update

### Behavior
* Skirts intersect a multi-body STL.
* Enable skirts and slice a multi-body STL (example included is the heat bed cable clamp from Prusa).
* No collision with the body.
* Skirt goes right through additional bodies.

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
[image]
[heatbed_cable_cover.zip](https://github.com/prusa3d/Slic3r/files/2045643/heatbed_cable_cover.zip)
[image]

BETLOG, 2018-05-29:
+1
I've seen this too. Unfortunately it's been a while, so I can't really contribute specifically to triage.

FidelCapo, 2018-05-30:
Thank you for the report. I looked at it and @BETLOG is right. Indeed, both parts of the object are not on the mat. When I split a multi-body object, the skirt is generated correctly.

zbrozek, 2018-05-30:
Huh, guess I'll go open an issue on the Prusa hardware then. Good eye - thank you!

FidelCapo, 2018-05-31:
I'm closing the issue. Feel free to reopen if necessary.

closed 2018-05-31 by FidelCapo