prusa3d/PrusaSlicer #928: Slic3r PE 1.40.0-beta - 0.05mm Gyroid@5% Overextrution Issue

TheZeroBeast, 2018-05-28:
### Version
Slic3r PE 1.40.0-beta
Running MK3 3.2.1 FW

### Operating system type + version
OS X 10.13 High Sierra

### Behavior
* _Infil is excessively overextruded resulting in print failure_
* _Steps needed to reproduce the problem_
  * _Load attached STL file_
  * _Using default options in Slic3r, select MK3, PrusaPLA,0.05mm ULTRADETAIL MK3_
    - _Change infill to Gyroid @ 5%_
  * _Print and observe the excessive overextrution on infill_

* _Expected Results_
  * _Uber fine detailed print of desired STL_
* _Actual Results_
  * _Parimeter extrution issues_
  * _Infill overextrusion_
  * _Print manually stopped due to continuously crashing into infill

#### STL/Config (.ZIP) where problem occurs
[PoopEmoji_Rev3 - STL](https://github.com/prusa3d/Slic3r/files/2046218/PoopEmoji_Rev3.zip) from https://www.thingiverse.com/thing:1023195
[PoopEmoji_Rev3 - Gcode](https://github.com/prusa3d/Slic3r/files/2046222/PoopEmoji_Rev3.Gcode.zip)
[image]
[image]
[image]
[image]

supermerill, 2018-05-31:
Can you link your profile archive?
file-> export config

TheZeroBeast, 2018-06-01:
Hi @supermerill,

I've attached the config as requested although this time looking through I think I may have had woodfill profile still on from the previous print, with the 120% extrusion. I'm going to reprint first few layers with new slice and confirm.

If infill doesn't need bridge flow that would also be good but may degrade the gyro infill if not properly formed. Do you have any photos from your prints with it off for gyriod?

[config.ini.zip](https://github.com/prusa3d/Slic3r/files/2062058/config.ini.zip)

supermerill, 2018-06-01:
No photos but i didn't see any major problem, even at low infill and ludicrous speed. People who want to be extra careful can reduce the infill speed anyway.

TheZeroBeast, 2018-06-02:
Still an issue, double checked settings where correct, printed and still failed.
new gcode attached.

[image]

[PoopEmoji_Rev3.gcode.zip](https://github.com/prusa3d/Slic3r/files/2064856/PoopEmoji_Rev3.gcode.zip)

supermerill, 2018-06-03:
Hum, the "use_bridge_flow" parameter is a hardcoded setting, you have to recompile slic3r to modify it.

bubnikv, 2018-06-04:
Do we want to disable the "use_bridge_flow"? This would mean to verify that
the gyroid infill is printable at the normal layer heights, and to modify
the fudge factor to the infill density.

bubnikv, 2018-06-04:
@lukasmatena will look into this issue.

bubnikv, 2018-06-12:
@lukasmatena tried to disable the bridge flow, and it certainly helps for the very thin layers. If we just disabled the bridge flow now, the gyroid infill may not be stable enough, so we will likely not do any change for the 1.40.0 release, but for the future releases we will likely adjust the behavior somehow.

TheZeroBeast, 2018-06-24:
Hey @lukasmatena @bubnikv, I tried with my 0.8mm nozzle and with 0.2mm layer heights and standard 20% gyroid infill and had the same issue resulting in a failed print. Going back to grid infill resolves the issue. Something has surely been missed with handling gyroid that I hope we're able to fix, it's such a great infill.

supermerill, 2018-06-26:
> Confirmed significant over-extrusion at both 0.2mm

 @scalablecory 
can you send me your settings and stl?

bubnikv, 2018-10-23:
The bridging flow was disabled for Gyroid infill with b429b39
The fix will be in 1.41.1 final.
Closing.

closed 2018-10-23 by bubnikv