prusa3d/PrusaSlicer #929: Random bridge infill on second layer causing heavy overextrusion

Itox001, 2018-05-29:
### Version
1.40 alpha

### Operating system type + version
Windows 7 Ultimate 64x

### Behavior
When slicing this model, there is a patch of bridge infill on the second layer, which is printed with a very heavy flow causing overextrusion. Here: 
[image]
Flow visualization shows the super heavy flow:
[image]

#### STL/Config (.ZIP) where problem occurs
Config and model:
[Evidence.zip](https://github.com/prusa3d/Slic3r/files/2046713/Evidence.zip)

dartrax, 2018-05-30:
I also had problems with "Bridge Infill". In my case, I print with FilaFlex, normally layer height is 0.8 and extrusion width 0.8, so the cross-section area is 0.5mm².
When "Bridge Infill" comes in, the extrusion height and width both go up to 1.1, which results to a cross-section area of 1.1mm². As a result, the volumetric flow gets more than doubled!

[image]
[image]
[image]

I had severals prints coming out like this before I realized what the problem was.

[image]

The way to solve this would probably be to limit the "Max Volumetric Speed"-setting, so slic3r will reduce printing speed.

<printer related stuff in case someone still has issues>
Unfortunately, while Gcode seems to be ok with this, in my special case the printer ignores the lower printing speed, there is some kind of "min speed". [There are hints](https://forum.repetier.com/discussion/3033/firmware-ignoring-speeds-lower-then-300mm-min) that the minimal printing speed of the repetier firmware <v1.0 depends on the maximum (!) acceleration limit, so if a firmware update is not an option, this should be the path to follow.
</printer related stuff>

bubnikv, 2018-06-04:
The problem seems to be the anchoring of bridges, which is excessive in this case. We shall certainly fix this, but unfortunately this will not be done in the 1.40.0 release.

dartrax, 2018-10-02:
Is this fixed or will this be fixed in 1.41.1 ?

mathiasvr, 2021-06-01:
I just had this problem now. The bridge infill has a much higher layer height than other feature types.

closed 2026-09-29 by prusaslicer-bot[bot]