prusa3d/PrusaSlicer #930: Some env variables do not return values

BETLOG, 2018-05-29:
### Version
Slic3rPE-1.40.0-beta.linux64-full-201805231653.AppImage

### Operating system type + version
cat /etc/issue
Ubuntu 14.04.5 LTS \n \l
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 14.04.5 LTS
Release:        14.04
Codename:       trusty
uname -a
Linux betlogbrick 3.13.0-142-generic #191-Ubuntu SMP Fri Feb 2 12:13:35 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

### Behavior
* _Describe the problem_
I use slicer environment variables in my output filenames. Some do not return a value
eg:

`[input_filename_base]__[first_layer_bed_temperature]-[bed_temperature]C_[ first_layer_temperature]-[temperature]C__[layer_height]mm-[min_layer_height]-[max_layer_height]mm__[perimeters]x[nozzle_diameter]-[perimeter_extrusion_width]-[external_perimeter_extrusion_width]mm-[external_perimeter_speed]mms__[fill_density][fill_pattern]__[timestamp].gcode`

`test__60-60C_200-200C__0.2mm-0.12-0.32mm__3x0.4-0-0mm-40mms__20%triangles__20180529-162305.gcode`

* _Steps needed to reproduce the problem_
Enter the following into the "Output filename format" field:
`[input_filename_base]__[external_perimeter_extrusion_width]-[external_perimeter_speed].gcode`
Drag a model into the platter, and hit "export G-code"
Note that :
[input_filename_base] - is populated and valid
but 
[external_perimeter_extrusion_width]
[external_perimeter_speed]
are always 0

* _Expected Results_
Even if the settings field is set to 0, the variables above should return their default numeric values.

* _Actual Results_
They always return 0
Do not return default values.
Do not return specified values when those fields are specified.

bubnikv, 2018-05-29:
> Even if the settings field is set to 0, the variables above should return their default numeric values.
> They always return 0

Yep, that is how it works now. It returns the configuration value you entered.

mizhgan, 2019-11-30:
This issue is still here at 2.1
What my settings are:
[image]
What my custom G-code values are (this is just for testing purpose):
[image]
What my generated gcode file are:
[image]

bubnikv, 2020-02-04:
Implemented with 7d7f66aca6e445ae235252af5f544ef7144e19e1

Please note that the old placeholder syntax [variable_name] still produces the old, unsubstituted result. If you put 50% there, you will get 50%.
On the other hand, the new placeholder syntax {variable_name} will produce numbers as close as possible to what the slicing back-end works with. There may be some discrepancies of extrusion width dependent on layer height if variable layer heigth is in action though.

Closing. It will be part of the upcoming PrusaSlicer 2.2.0-alpha4.

closed 2020-02-04 by bubnikv