prusa3d/PrusaSlicer #932: solid infill near walls

Zawadaki, 2018-05-29:
How to get rid of this solid infill near walls especially sloping one, I disabled option to settings, but instead to get less or none I get more !, how to completely disable this ?
[image]

bubnikv, 2018-05-29:
There is currently no way to disable this kind of infill. The purpose is to
"anchor" the top layers into the sparse infill, but I agree that this is
mostly too much and it is not controllable. We should at least add some
parameter.

This issue is a duplicate of https://github.com/prusa3d/Slic3r/issues/668.
Closing this one.

closed 2018-05-29 by bubnikv

Zawadaki, 2018-05-29:
additionally one perimeter would be sufficient or just print this kind of infill very slow to give material time to cool down