prusa3d/PrusaSlicer #933: [REQUEST] Name of profiles (print, filament and printer] used at the top of the Gcode

Dibujor, 2018-05-29:
### Version
_Slic3r 1.40 Beta
For calibrating and trying purposes it would be neat having the name of the profiles used on that gcode file , as comments, somewhere at the top of the Gcode file. 

Somewhere there under "generated by Slic3r Prusa Edition [...]".

I'm new to 3D printing and I'm doing so much proofs that sometimes I lose my mind trying to remember.

Yeah, I know you can add that info in the notes section of each profile, but then you have to look for it at the bottom of the file among A LOT of code lines....
[image]

bubnikv, 2018-06-04:
We will not do that for 1.40.0 final release, but we should do that in some later version. Namely, there is an empty unused print_id, filament_id and printer_id variable stored in the profiles and a G-code already, which we should populate with the file name if it is empty.

Dibujor, 2018-06-04:
That will make life so much easier when you have a lot of profiles and want to remember at a glance which settings did you use to slice that piece...

Thanks!

sarf2k4, 2018-06-21:
Hi,

I requested a new request in the firmware section that may be used inside slic3r as well at https://github.com/prusa3d/Prusa-Firmware/issues/855

Reason being is that I wanted to point out we might really need prompt to print with info screen

bubnikv, 2018-06-28:
On the branch amf_activate_existing_presets, the "print_settings_id", "filament_settings_id", "printer_settings_id" keys are stored at the end of the G-code together with the rest of the parameters.

The branch will most likely be merged into 1.40.2 release, which is due in around 2 to 3 weeks.

Dibujor, 2018-06-28:
That's cool! Thanks bubnikv

bubnikv, 2018-08-02:
The information is there, altough not at the top of the file. 
Also our latest alpha profiles configure the G-code output file name to contain the layer height and filament type.
Closing.

closed 2018-08-02 by bubnikv