prusa3d/PrusaSlicer #934: [Request]

mvasilakis, 2018-05-29:
### Version
_1.40.0 beta_

### Operating system type + version
_Mac OS 10.13.4_

### Behavior
* _Describe the problem_
Slicer PE 1.40.0 beta introduced an update system to keep Prusa profiles updated. It would be nice if a GUI method was added to keep other profiles up to date as well. If we could add a link to say dropbox for our own profiles and when the dropbox profile/bundle is changed/updated/added it would auto update on all our slic3rs across platforms and computers. Also when someone like Chris Warkocki changes or adds  profiles they would also auto updated in Slic3r. 

* _Expected Results_
A section in preferences for adding links to profiles and/or bundles online. 

_Is this a new feature request?_
Yes

bubnikv, 2018-05-29:
This should be working already to some extent. We support alternate vendors.

vojtechkral, 2018-05-29:
@mvasilakis It is implemented, but there's isn't a public documentation yet on this. I have a draft documentation on how to do this around, but it's internal at this point and needs some polishing. I suppose we could publish it as in wiki here when the time is right,

vojtechkral, 2018-05-29:
@mvasilakis Well, I'll share my WIP documentation to spare you the trouble. Feedback would be valuable to us anyway, but since it's WIP, take it with a pinch of salt.

[Vendorbundlesandupdatingprocess-290518-1827-18.pdf](https://github.com/prusa3d/Slic3r/files/2049218/Vendorbundlesandupdatingprocess-290518-1827-18.pdf)

mvasilakis, 2018-05-29:
Thanks. I read through it quickly. Interesting set up. I'll mess with it and get back to you.

bubnikv, 2018-12-11:
Any update on your progress?

bubnikv, 2019-01-29:
No response, closing.

closed 2019-01-29 by bubnikv

mvasilakis, 2019-04-15:
Sorry. Life got in the way and this had to get benched. I'll report back if/when I can get to it.