prusa3d/PrusaSlicer #936: Prints with X axis over about 1074mm fail

lmcnjmiller, 2018-05-29:
### Version
1.39.2+win64

### Operating system type + version
Windows 10

### Behavior
Objects with x axis over 1074mm do not slice correctly or at all. 
Complex objects will only slice features under this length. 

To reproduce simply place an object on the bed that exceeds 1074mm and slice. 

Here I sliced a cube that is 1074x10.74x10.74

[image]

Same cube slightly smaller. 

[image]

#### STL/Config (.ZIP) where problem occurs
Any mk2 or other config is effected. 
the stl is just a cube. 
If you are wondering I do have a printer this big.

FidelCapo, 2018-05-30:
Thank you for report. I tested this issue in Slic3r 1.39.2 and 1.40.0-beta. In beta, this issue is probably fixed. Can you please try this issue in Slic3r 1.40.0-beta? Thank you

FidelCapo, 2018-05-30:
Great. I'm closing the issue. Feel free to reopen if necessary.

lmcnjmiller, 2018-05-30:
Spoke to soon. Objects now print with Solid infill. 

Object Long on X axis:
[image]

Object rotated 90 degrees to be long in Y axis:
[image]

lmcnjmiller, 2018-06-04:
Here is the AMF
[1500mm.zip.amf.zip](https://github.com/prusa3d/Slic3r/files/2068996/1500mm.zip.amf.zip)

lmcnjmiller, 2018-07-10:
This issue is now resolved in the latest version. 1.40.1

FidelCapo, 2018-07-12:
Glad to hear it. So can I close this issue now?

lmcnjmiller, 2018-07-24:
I would say so.

closed 2018-07-24 by vojtechkral