prusa3d/PrusaSlicer #937: variable layer anomaly

BETLOG, 2018-05-30:
### Version
Slic3rPE-1.40.0-beta.linux64-full-201805231653.AppImage

### Operating system type + version
Kubuntu 14.04.5

### Behavior
* _Describe the problem_
Using variable layer edits on multi part print does not print some objects

* _Steps needed to reproduce the problem_
Import miltipart stl, select 'split, manually arrange and rotate.
Select part and assign some variable layer thickness; reducing from normal thickness to minimum. - these work ok
Select part and make entire object minimum variable layer thickness, and then edit in some max layer thickness sections - this part renders as preview type: 'shell', but does not print.

* _Expected Results_

* _Actual Results_
[image]
[image]
[image]
[image]
[image]
[image]
[image]

FidelCapo, 2018-06-01:
Unfortunately, I am not able to reproduce the issue you described. Can you please provide AMF or 3MF files where the issue is occurs? Thank you

BETLOG, 2018-06-01:
Unfortunately I am also unable to reproduce it, and have overwritten the stl several times since then.
At the time I was playing with the new display features, so it's probable i did something unusual that caused it. Normal usage does not seem to be a problem.
I'll report if i do manage to reproduce or otherwise encounter this again.

BETLOG, 2018-06-04:
Sure. I'll report back if i encounter it again, and attempt to figure out how to reproduce.

bubnikv, 2018-06-04:
Cannot reproduce, closing for now. Please reopen if the problem reoccurs.

closed 2018-06-04 by bubnikv