prusa3d/PrusaSlicer #938: Automatically vary infill density

Looxoor, 2018-05-30:
I came across this recently and couldn't find an issue here that addresses this request.
[Makerbot's MinFill](https://www.makerbot.com/media-center/2017/03/20/introducing-makerbot-minfill)   looks to me like a very interesting concept.

Is it something that can be implemented in Slic3r?

_Is this a new feature request?_
Yes

supermerill, 2018-07-04:
If you run on windows, you can grab my experimental release in the pull request #821 (second post)

If you want to compile, the easiest is to grab my master on https://github.com/supermerill/Slic3r

rongith, 2018-07-07:
@supermerill Thank you. I compile now from https://github.com/supermerill/Slic3r. Nice bells and whistles. I will report there for issues.

rtyr, 2021-01-21:
You may try new Adaptive Cubic and Support Cubic infill types added in PS 2.3.0.

closed 2026-09-29 by prusaslicer-bot[bot]