prusa3d/PrusaSlicer #939: [Request] Better auto arrange

ManuGithubSteam, 2018-05-31:
### Version
_Version of Slic3r used goes here_
1.39
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Linux

### Behavior

Put some stuff on the platter who will fit the platter with some iteligent arangeing and then try to arrange

  * _Screenshots from __*Slic3r*__ preview are preferred_
How it is:
[image]

How it should be:
[image]

_Is this a new feature request?_
YES

#### Request: Make the auto arrrange more inteligent and rotate stuff or dont make the distance to wide so it will fit on the platter.

BETLOG, 2018-06-01:
+1
There is some obvious (and probably very easy to implement) room for improvement here.
PS: I'd have put the four circles beside the long parts and centred them all
PPS: is there a request for "ability to select and move all objects at once (in 2d view)"?

vojtechkral, 2018-06-01:
A new arrange algorithm is being implemented by @tamasmeszaros , I believe there are promising results already, but not yet part of current release...

rongith, 2018-06-02:
https://github.com/Rhoban/Plater do a good job

bubnikv, 2018-08-02:
New auto placement is implemented in SLic3r 1.41.0-alpha3. Closing.

closed 2018-08-02 by bubnikv