prusa3d/PrusaSlicer #940: wipe tower first layer grid fill far too fast

berndu, 2018-05-31:
### Version
1.40.1 beta

### Operating system type + version
Max OS

### Behavior
There is a reset of the speed in gcode between wipe tower brim and wipe tower first layer grid fill.

Generated gcode looks like this..

G1 X157.302 Y71.599 F7000
G1 X97.302 Y71.599
G1 X157.302 Y71.599
; CP WIPE TOWER FIRST LAYER BRIM END
;-----------------------------------
**G1 X97.802 Y57.699 F7200**   <--- bad?
;--------------------
; CP EMPTY GRID START
; layer #1
G1 X97.302
G1 Y57.199
G1 X157.302 Y57.199 E3.2600

This causes the first layer to get print very fast - the first layer get ripples and get ripped off on the next layer.

In the code this is cause by line WipeTowerPrusaMM.cpp:1178 - there the speed is hard coded.
So as a user i have no chance to change this on generating the gcode file.

Please make it adjustable or slower... maybe it can be calculated from another setting.

lukasmatena, 2018-05-31:
Thanks for the report. This issue is already fixed by https://github.com/prusa3d/Slic3r/commit/e4aff5b08a9e6ac4adb4c70515aa28e607a1b822, which is already in the current master and will be part of the next release (you can also compile it yourself before that). The speed was significantly decreased - the F7200 is a travel feedrate, which was not changed before printing of the fill.

spiff123, 2018-05-31:
Great! (sorry posting with other account)
I will try that!

bubnikv, 2018-06-04:
Will be fixed in the release. Closing.

closed 2018-06-04 by bubnikv