prusa3d/PrusaSlicer #941: Slight layer shift on every 5th layer when slicing some models via Slic3r

devilhunter84, 2018-05-31:
### Version
1.39.2 and 1.40-beta

### Operating system type + version
Windows 10 fall update

### Behavior
On some models like a cube or something with a flat surface on the X axis Slic3r PE does a slight layer shift every 5 layers, resulting in a very broken surface finish.

Using stock Slic3r MK3 profiles and stock Cura MK3 profiles

Rotating the model sometimes fixes the problem.

I've uploaded pictures of the results here:
https://imgur.com/a/sQT0hXb

printed 3d Object is a simple teardrop, 1 shell, no top/bottom layers, no infill

Pictures are:

1st 3 pics: Cura with the flat side to the front
results: perfect x side, perfect bend, perfect pointy end
2nd 3 pics: Slic3rPE 1.39 and 1.40-beta with flat side to the front
results: layer shifting on every 5th layer slightly to the back, slightly broken bend, perfect pointy end
3rd 3 pics: Slic3rPE 1.39 and 1.40-beta with model rotated 90 degrees, so that the flat side is to the right
results: perfect x side, perfect bend, perfect pointy end
4rth 3 pics: Slic3rPE 1.39 and 1.40-beta with model rotated 90 degrees to the right and 'wipe while retracting' disabled
results: perfect x side, perfect bend, perfect pointy end
5th 3 pics: 3rPE 1.39 and 1.40-beta with the flat side to the front and 'wipe while retracting' disabled
result: layer shifting on every 5th layer slightly to the back, slightly broken bend, perfect pointy end

Included are:
gcodes, STL and pictures

[teardrop.zip](https://github.com/prusa3d/Slic3r/files/2058897/teardrop.zip)
[gcodes.zip](https://github.com/prusa3d/Slic3r/files/2058898/gcodes.zip)
Pictures https://imgur.com/a/sQT0hXb

stahlfabrik, 2018-06-29:
That looks to a noob like me like a contribution to #602
Interesting that Cura generates a even print quality while slic3r seems to have an issue. 

What is Prusa Research‘s conclusion what is happening here?

bubnikv, 2019-03-12:
We have printed the teardrop model. The thing with the model is, the thin wall only fits a single line, therefore it is sliced using a gap fill algorithm, which produces an ugly seam in this particular case.

rtyr, 2020-04-15:
Can you try again with current version of PrusaSlicer (2.2.0)? 

I looked at the gcode and it seems to me, that those lines (shifts) are actually caused by slight overextrusion at the beginning of the layer. Seam position placement has some pattern there, which is repeating itself every 5 layers.

[image]

Anyway, I resliced it with current PS 2.2.0 using exactly same settings and the result is now different (and the "jiggle" extrusion effect at the start of the layer is gone). I even printed it on the MK3 and it looks fine.
I guess we can close this issue.

closed 2020-04-16 by rtyr