prusa3d/PrusaSlicer #942: Feature request: override cooling fan speed for top infill

DrLex0, 2018-05-31:
### Version
1.40.0-beta+

### Operating system type + version
Any

### Behavior
I would like to be able to enforce a specific fan speed (typically much lower or just off) while top infill is being printed and perhaps also for parts of the next layer printed on top of this infill. Two reasons:
* Top infill is usually printed much slower and at a narrower extrusion width than the rest to obtain a nice finish. This means that the cooling fan also spends much more time blowing air over the same areas which can become cooled too much or too quickly. This can lead to an ugly dull top surface for filaments that normally have a glossy finish (especially PETG).
* Layer adhesion between this top infill layer and the next one can be affected if the next layer is being printed on the under-cooled lower layer. This gets worse with auto cooling, if the next layer is short hence fan speed is boosted (see images for such situation). I'm not sure what would be the best approach here. The low speed could be maintained for anything printed directly onto top infill, or there should be a delay when boosting fan speed for a short-time layer following a long-time layer, or maybe both. Maybe a more general solution is needed to ensure that increased cooling for short layers won't degrade layer adhesion. 

[image]
[image]

The rightmost part in the photo was printed with manually reduced fan speed for those two layers, and the result is much stronger.

_Is this a new feature request?_
Yes!

supermerill, 2018-06-04:
Can i have your test model (stl) & settings ?

your test was made with pla or petg?

DrLex0, 2018-06-24:
Here is an example with PETG and how cooling affects the look of the top surface: [PETG-test.zip](https://github.com/prusa3d/Slic3r/files/2131174/PETG-test.zip)
In the photo below, the right model was printed as-is with the included settings. The left model was printed with the exact same G-code but with an extra ‘fan off’ command manually inserted before the top infill started printing.

[image]
Perhaps the cooled top surface looks better in this case because this was only a quick test print with few top layers hence the uncooled top surface has deformed from the heat. The point of this test is to show that the uncooled surface is much more shiny, and (not visible in the photo) more transparent as well.

(By the way, the ugly line through the top surfaces is another illustration of issue #471 (and the similar issue in mainline Slic3r) where infill in elliptical shapes is pretty much always marred by a travel move due to greedy path planning.)

DrLex0, 2018-06-24:
Here is an example with ABS, the model shown in the first photo and screenshot above. Here, too much cooling of the small top surface at 1.85 mm and the following short layer where the long shaft begins, causes a weak point (as shown in the photo): [ABS-test.zip](https://github.com/prusa3d/Slic3r/files/2131205/ABS-test.zip)
Yet, for the other layers the fan must blow at high speed to prevent the shaft from turning into a molten blob.

This is a bit tricky because the object is so small that the layers will only take a few seconds and the fan will always be at maximum speed. This would mean that this top layer fan override should even override the auto-cooling speed to prevent this kind of problem.

In this case I manually inserted a much lower fan speed for layers Z=1.85 and 1.95 and the result is so strong that attempts to destroy this print resulted in other parts breaking before this joint gave up.

supermerill, 2018-06-27:
I re-created this option in my branch (i created it some months before but removed it because i din't see any use for it after testing)
You can find a compiled version here #821
If needed, i can create a separate pullrequest.

OffensiveBias-08-145, 2024-03-22:
@supermerill Having a slicer setting that turns off the fan for top layers would be fantastic!

This helps with maintaining a gloss top layer with certain filaments 
(Ex: Polymaker Black PETG and Prusament PETG Jet Black)

Right now I am either having to watch the print and modify the fan speed both off and then back on again for top layers, or try and find specific layer heights and modify the GCode. 

Both are tedious and annoying. 

Since we are already able to set print speeds for certain layers, why not the same for the fan?

closed 2026-09-29 by prusaslicer-bot[bot]