prusa3d/PrusaSlicer #944: "unknown config option" with legacy .ini files

RadianM, 2018-06-01:
Slic3r 1.39.1-beta on Windows Version10.0.17134 Build 17134
"unknown config option" error when Importing config bundle from legacy versions of Slic3r (e.g. 1.2.9)

Not necessary to be regarded as a bug, but a feature request would be to add the ID of the option(s) that are throwing this error to facilitate the manual editing of legacy config files.
This kind of error has previously been reported here:
https://github.com/prusa3d/Slic3r/issues/795
https://github.com/prusa3d/Slic3r/issues/754

1.39.1-beta has a fix for Windows 10 updates that appear to have broken Intel graphics therefore I'm having to migrate to this prusa3d version and I am looking for the optimal way to transfer my numerous configs. Knowing the actual option(s) that are unrecognised might smooth this process - or is that unlikely?

bubnikv, 2018-06-04:
Did you try the latest beta?
https://github.com/prusa3d/Slic3r/releases/tag/untagged-485e220465fc98722917

Isn't your issue similar to https://github.com/prusa3d/Slic3r/issues/896 ? This has been fixed in the beta referenced above.

RadianM, 2018-06-05:
@bubnikv Dead link - sorry but there are many branches of this code and I'm unsure as to which might be the latest beta, and whether it is available pre-built. I listed the version I'm using in my first post above.

eowindel, 2018-06-05:
https://github.com/prusa3d/Slic3r/releases

bubnikv, 2018-06-11:
Is it working for you now? Can we close this issue?

bubnikv, 2018-12-11:
No response, closing.

closed 2018-12-11 by bubnikv