prusa3d/PrusaSlicer #947: Bridge start/end points in mid air

martin-git, 2018-06-02:
### Version
1.39.1-prusa3d-win64

### Operating system type + version
win10

### Behavior
Starting/End points of bridging lines are in mid air, resulting in failed bridges.

So I was testing different bridge extrusion multiplier and bridge speed settings for polycarbonate filament on a little test model and almost all failed rather randomly until i saw the path start/end points in the octopi gcode-viewer:

[image]

All start/end points circled yellow are mid air, causing the line to fail! So unless there is a hidden setting I don't know of which forces bridging paths to start above something solid, that shouldn´t happen. 

#### STL/Config (.ZIP) where problem occurs
[testbridge.stl.zip](https://github.com/prusa3d/Slic3r/files/2065332/testbridge.stl.zip)
[config.ini.zip](https://github.com/prusa3d/Slic3r/files/2065340/config.ini.zip)

bubnikv, 2018-06-04:
In your config you have "detect bridging perimeters" disabled, therefore the bridging logic does not apply.

bubnikv, 2018-06-11:
Is the explanation sufficient? Can we close this issue?

Petclaud, 2018-06-11:
Sorry, this is not a real response to this matter, but more a question related to it: I am just curious why 'detect bridging perimeters' is not ON by default in the provided profiles...?

bubnikv, 2018-06-11:
The "detect bridging perimeters" settings will generate thick lines at steep overhangs as well. That's why it is disabled by default.

Petclaud, 2018-06-11:
Ah ok great, thx for the super fast reply. Is it then a suggestion to mention this in the tooltip...?

maximlevitsky, 2018-06-30:
"The "detect bridging perimeters" settings will generate thick lines at steep overhangs as well. That's why it is disabled by default."
Sorry for the noob question, but what is wrong with that? Could you explain?

rennne, 2019-02-02:
If with "detect bridging perimeters" switched on the problem is still there:

The gray blob in the bridge, is a retraction:

[image]

rtyr, 2020-05-04:
I believe we can close this, the reason of described behavior was explained. The last screenshot is probably retraction after wipe movement.

closed 2020-05-04 by rtyr