prusa3d/PrusaSlicer #948: Installation program seem to overwrite your current print & printer settings

MrM40, 2018-06-04:
### Version
1.39.2

### Operating system type + version
Win7 x64

### Behavior
I had a older version on my system and then installed 1.39.2 to upgrade it.
(Well, I was expecting to get the option to specify installation folder in the installation program, but that was not possible...feature request)
But it seems the installation program deleted all my printer and print settings in "%APPDATA%\Slic3rPE\print" & "%APPDATA%\Slic3rPE\printer" :-(

bubnikv, 2018-06-04:
How did you install Slic3r? With the Prusa3D drivers install package?

MrM40, 2018-06-04:
My mistake
My settings was located in the folder specified with the "**--DateDir**" startup parameter.
In my case my old settings were located in
"C:\Users\Public\Documents\Prusa3D\Slic3r settings MK2 180"
:-)
The new located set the the installation program was "C:\Users\Public\Documents\Prusa3D\Slic3r settings MK2-MK2S-MK2.5 216"

closed 2018-06-04 by MrM40