prusa3d/PrusaSlicer #949: USB print: "Error:Line Number is not Last Line Number+1, Last Line: 0"

MrM40, 2018-06-04:
### Version
1.39.2+win64

### Operating system type + version
Win7 x64

### Printer
MK2S

### Behavior
Cannot control/print from Slic3r PE using USB.
I just updated to firmware **3.1.0**
When connecting to the printer in Controller I get this error:
-------------------------------------------------------------------------------------------
echo:echo: Last Updated: Nov 12 2017 11:07:38 | Author: (none, default config)
Compiled: Nov 12 2017
echo: Free Memory: 2158  PlannerBufferBytes: 1360
echo:Hardcoded Default Settings Loaded
echo:SD init fail
**Error:Line Number is not Last Line Number+1, Last Line: 0**
-------------------------------------------------------------------------------------------

MrM40, 2018-06-04:
It seem 3.1.0 it not the newest firmware, although it show as the newest on https://www.prusa3d.com/drivers/
Just upgraded to V3.2.1 and that just bricked my MK2S...DAMN!

vojtechkral, 2018-06-04:
@MrM40 That Line number error is pretty much just synchronization between slic3r and the pritner, it's not an indication of a problem unless it happens way too often. (And yes, the Controller tab code is in a need of some maintenance.)

> Just upgraded to V3.2.1 and that just bricked my MK2S...DAMN!
Are you positive you flashed the right variant of the firmware? I don't believe there is a firmware newer than 3.1.0 for the MK2S. Can you flash 3.1.0 onto your printer?

MrM40, 2018-06-04:
Well, I just went to https://www.prusa3d.com/drivers/, selected my printer (mks2).
And as 3.1.0 was not working directly from slic3rPE and it's rather "old", and that this problem seem to be a know problem (got a few hit when search for the error on google) I expected you already fixed the problem.
I clicked  "Older versions" to downgrade, but to my surprise there was a newer version 3.2.1 from 18/5-2018. The flashing went find according to the flash tool saying "Firmware updated Successfully!" but for sure the printer was NOT fine. It's totally bricked now. I fear I need a now board for 130$ :-(

3Delights, 2018-06-04:
You should be able re-flash it with version 3.1.0.  The latest version for the MK2 and the MK2S is 3.1.0.  The board will not necessarily seem alive and the LCD display may be blank or corrupted but it should still flash back to the older version.  Give it a go.

When you click on the "Older versions" link on the Prusa website it takes you to a list of _every_ previous version for every printer in a single list.  Next to the version number is a label indicating which model that version is compatible with.

They plan on eventually rolling all the firmware for all the models into one, but have not yet done so.

MrM40, 2018-06-04:
It sure did work to flash it back……THANKS J

bubnikv, 2018-06-11:
Isn't this a firmware issue? Should we close it here in the Slic3r issue tracker?

MrM40, 2018-06-11:
All I can say it is you can fine control the printer from Pronterface using the USB interface. E.g. moving the printhead x, y and z.  But it doesn't work in Slic3r PE.
Didn't try to print.

closed 2021-01-18 by rtyr