prusa3d/PrusaSlicer #950: Color print does not work properly with gcode generated in sli3r "1.39" or "1.40 Beta"

Dainse, 2018-06-04:
### Version
1.39 /1.40 beta

### Operating system type + version
Windows10 Pro 64bits

### Behavior

Color print does not work properly with gcode generated in sli3r "1.39" or "1.40 Beta" using the official Mk3 Profiles.

I tested with the web version of colorprint ( https://www.prusaprinters.org/color-print/ ) and with the version app in Windows 10 pro 64 bits ( http://prusa3d.com/downloads/others/ColorPrintbyPrusa3D. zip # _ga = 2.102140097.1388865899.1528117604-1208716770.1527630374 )

Dragging the gcode to the application freezes and does not allow adding pauses for color changes.

Some help? Issue?

[Redimensionado Llavero.zip](https://github.com/prusa3d/Slic3r/files/2068421/Redimensionado.Llavero.zip)

chadwixk, 2018-06-05:
Thought it was just me with 1.40.0 beta win64 ...same issue here. Is there any work around...or how do you manually modify the gcode...or maybe I just need to go back down to 1.38?

bubnikv, 2018-06-06:
We are testing a solution.

bubnikv, 2018-06-06:
Please test the web color print again. The fix should have been deployed. Thanks.

chadwixk, 2018-06-06:
Thanks, colorprint is now working for me!

Dainse, 2018-06-06:
The web version has been corrected! works correctly with the same Gcode that previously caused an error.
Will there be an update of the desktop application?

docmarenga, 2018-08-02:
I´ve got a crash when try load de gcode en ColorPrint.  "Uncaught node.js Error"
some days back, work without problems.  i try with different gcodes and the same error .
[image]

Dainse, 2018-08-02:
Use the web version. Its updated. 
https://www.prusaprinters.org/color-print/

Good luck!

docmarenga, 2018-08-02:
Thanks very much!!!!!!!

Saludos Rodrigo Marin