prusa3d/PrusaSlicer #952: Plater tab crashes on ubuntu

subsite, 2018-06-05:
### Version
1.40.0-beta+linux64 (Run from AppImage `Slic3rPE-1.40.0-beta.linux64-full-201805231653.AppImage`)

### Operating system type + version
Ubuntu 18.04 

### Behavior
* _Describe the problem_
Control panels aren't redrawn when switching back to the plater tab (see screenshot). It still responds to clicks if you click the area the button normally is. This problem first appeared with 1.40.0-beta, 1.40.0-alpha still works normally.

When run from the terminal, it gives the following output (The `IA__gtk_color_button_set_color`-error displays immediately on startup, the `IA__gdk_window_thaw_updates` appears when the bug is triggered):
```
$ ./Slic3rPE-1.40.0-beta.linux64-full-201805231653.AppImage 
Creating main frame...

(slic3r.pl:24539): Gtk-CRITICAL **: 18:32:24.961: IA__gtk_color_button_set_color: assertion 'color != NULL' failed

(slic3r.pl:24539): Gdk-CRITICAL **: 18:32:34.314: IA__gdk_window_thaw_updates: assertion 'impl_window->update_freeze_count > 0' failed

(slic3r.pl:24539): Gdk-CRITICAL **: 18:32:34.314: IA__gdk_window_thaw_updates: assertion 'impl_window->update_freeze_count > 0' failed

(slic3r.pl:24539): Gdk-CRITICAL **: 18:32:34.314: IA__gdk_window_thaw_updates: assertion 'impl_window->update_freeze_count > 0' failed

(slic3r.pl:24539): Gdk-CRITICAL **: 18:32:34.314: IA__gdk_window_thaw_updates: assertion 'impl_window->update_freeze_count > 0' failed

# ...and the same error repeated numerous times
```

* _Steps needed to reproduce the problem_
This happens after changing some parameters in another tab and then trying to return to the Plater. For me, this can be reproduced by opening the Filament tab and then selecting one of the sub menu items (e.g. Cooling).

[image]

[config.ini.zip](https://github.com/prusa3d/Slic3r/files/2073027/config.ini.zip)

lukasmatena, 2018-06-05:
Hi. You might wanna have a look at https://github.com/prusa3d/Slic3r/issues/898

Please, report if that helped to solve the problem.

subsite, 2018-06-05:
Thanks, it seems to be the same problem. However, the mentioned workaround doesn't apply any more, as Ubuntu is now stable at 18.04 Bionic and is no longer using Unity as its desktop environment (they moved back to Gnome). The tweak to disable overlay scrollbars doesn't seem to be available in the tweak tool any more.

I will keep looking for a similar workaround for 18.04, but it would be nice if Slic3r were usable without changing the look and feel of the whole operating system. I'll have to move back to 1.39, but I'll really miss all the nice features of 1.40.

subsite, 2018-06-05:
Ok, I might have found a similar workaround for 18.04.

In a terminal, enter `gsettings set com.canonical.desktop.interface scrollbar-mode normal`

I added this to the https://github.com/prusa3d/Slic3r/wiki/Known-Issues-and-Solutions

lukasmatena, 2018-06-05:
Awesome, thank you very much. Since you found a way to fix it and there is apparently nothing else we can do about it (if the bug is in Ubuntu itself), I will close the issue, if you don't mind. If you do, reopen it at will. Thanks again for your feedback.

closed 2018-06-05 by lukasmatena

subsite, 2018-06-05:
No problem, we can close this. Thanks!

bubnikv, 2018-06-06:
Thanks guys for your help.