prusa3d/PrusaSlicer #953: 1.40.0 Beta: Missing information on preview platter after slicing

Boby71, 2018-06-05:
### Version
1.40.0 Beta

### Operating system type + version
Win10 64 Bit

### Behavior
Not sure if it's a bug or a feature request, but then I did a slicing of an object, in the preview the information of the view/show (e.g. feature type) and the print information (cost, duration,...) is NOT shown in the preview. I need to export(!) the gcode to get the additional (enhanced) information.

In my opinion, this is not a really smooth workflow because I need to save a gcode I probably don't need/want, because I discovered a problem in the preview and adjust my print settings.

I see two possible options to improve this behavior:
1) automatically generate the additional information or
2) add a small button right beside the view/show dropdowns to generate the enhanced information (if it's not being done automatically because of resource/time issues).

If it's really a resource issue, then leave the decision to the user - but only offering this information after exporting the gcode is not logical (from user perspective) - thanks!

[image]

[image]

lukasmatena, 2018-06-05:
Hi. We know this needs improvement and intend to do so according to your first suggestion (by generating the gcode in the background) . This should be included in 1.41, as already mentioned in 1.40-alpha change log. It will require some extra work that we didn't have time to do yet, there is no other reason it is this way.

bubnikv, 2018-06-11:
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/800
Closing.

closed 2018-06-11 by bubnikv