prusa3d/PrusaSlicer #955: Feature Request : Dark mode for macOS Mojave

Tucker2015, 2018-06-06:
### Version
1.4.0-beta

### Operating system type + version
macOS Mojave Beta

### Behaviour

Would like a Dark Mode to integrate with the new mode in macOS Mojave when it is released.  

_Is this a new feature request?_

Yes

bubnikv, 2018-06-06:
Thanks for the report. It seems we are now having similar issues to what we dealt with on Linux in regard to dark modes. We need to get the Mojave machine.

CristianLeonRO, 2018-06-19:
It would help alot a Slic3r PE dark mode also on Windows.

tracernz, 2018-06-19:
The tab controls on all Slic3r windows are completely black on Mojave with dark mode, the buttons and text cannot be seen.

bubnikv, 2018-07-05:
There are rumors, that the upcoming beta update of OSX Mojave fixes this issue.

pointbrink, 2018-07-07:
rumors were true.. updated mohave tonight (18A326h) and my tabs are visible again

FidelCapo, 2018-07-09:
So I close this issue. If the problem occurred again feel free to open it again.

closed 2018-07-09 by FidelCapo