prusa3d/PrusaSlicer #957: Support layers not in sync with object layers?

chadwixk, 2018-06-07:
### Version
1.40.0 beta win64

### Operating system type + version
Windows 10 64B

### Behavior
Not sure if this is a problem or just something I don't understand. I was having issues with layer 2 of the support areas not adhering to layer 1 of support. After looking at the layers,  I see that the support layers seem to be further apart than the object layers. The print is set to .20 mm layers, but the support layers are printing at .20, then next at 0.44 instead of 0.40...which may explain why there is an adhesion issue between the support layers as they are slightly further apart.

Are there settings that control this? I see the "Synchronize with object layers" setting that is greyed out, but it seems to imply that is only useful for multi-extruder setups, which I don't have. But otherwise implying that the object and support layers are normally NOT printing at the same layer heights...why?

Just trying to learn. Thanks in advance. Here's some pics and the zip file of stl, configs, and gcode
[image]
[image]
[image]
[image]
[image]
[image]
[Slic3r.zip](https://github.com/prusa3d/Slic3r/files/2078970/Slic3r.zip)
.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-06-07:
This is not a bug, this is a feature. The support layers are not
synchronized with the object layers for two reasons.

First, this allows the support layers to grow faster than the object layers.

Second, having the support layers independent allows Slic3r to follow the
gap between the object and the support top surface exactly. Other slicers
usually allow the gap to be defined in multiples of object layers only.

chadwixk, 2018-06-07:
Gotcha. Ok thank you for the explanation!

closed 2018-06-07 by chadwixk

Lanzentraeger, 2020-06-19:
If it is a Feature, it works not good. If i want to print Support with 130mms, it slows down to 10mms, because of the minimum Layer Time. It only works, if i set it to 0. It is logic, because the Layer have to be same hight. But not, if i want to print with for example Layer hight 0.16 and also Support with 0.16. Why it ist not possible, to set the Option sychronize with object layers then?