prusa3d/PrusaSlicer #958: 1.40.0 Beta - Crash (due to big STL?)

gdombiak, 2018-06-07:
### Version
Using latest 1.40.0 Beta

### Operating system type + version
Ubuntu 16.04. Running AppImage.

### Behavior
Do not how to reproduce but here is what I did and the logs I found. Let me know if more info is needed and how to get it.

1. Loaded STL (too big to upload here)
1. Loaded Part (this is tree support generated from meshmixer)
1. Sliced and generated gcode
1. Went to preview and started to rotate image by using mouse
1. Kaboom. App crashed and closed

Here is the info I got from syslogs:
[image]

I think that these log entries are when the crash happened. I started Slic3r again, repeated steps and it worked this time.

_Is this a new feature request?_ NO

#### STL/Config (.ZIP) where problem occurs
It is 76MB so cannot upload here. Happy to share by email.

bubnikv, 2018-06-07:
Please share the STL through a drop box or any other public sharing service.

The 3D path preview is very memory intensive, so it is quite possible that you ran out of memory.

bubnikv, 2018-06-07:
By the way, does the Slic3r 1.39.2 slice and display the 3D paths correctly?
Thanks for the test.

gdombiak, 2018-06-08:
I tried to reproduce it again with 1.40.0 Beta and ... of course... it worked fine. Tried to find a core dump but there was none. No idea how or why the app crashed and was closed.

I left the zipped file [here](https://www.dropbox.com/s/ok2le7ezy7ihylt/alien.zip?dl=0)

Loaded the same STLs on my 1.39.2+ (on my mac), sliced, went to preview and also generated g-code. Things appeared fine in preview. Not sure what you mean by "display the 3D paths". How do I test that? :)

PS: It is so painful to go back to 1.39.2 after tasting the sweet honeys of 1.40.0. :) There is no going back now. Great job guys. I will open a new issue with suggestions for the preview on 1.40.0 for things I usually do/need and would be nice for slic3r to implement. Low hanging fruit really.

gdombiak, 2018-06-09:
And .... it happened again at some random place while manipulating the STL. App just crashed. This is all I found in the system logs at the time of the crash:

[image]

bubnikv, 2018-08-02:
As I understand it, you cannot reproduce the crash, right?
There are some issues regarding the cancellation and restart of the slicing process, but we are working on it.

gdombiak, 2018-08-02:
Hi @bubnikv 

I moved back to my Mac (my main computer) and Slic3r works great there. I haven't been using Slic3r on my old notebook with Ubuntu except to run the 1.41 alpha 2 to flash latest firmware 3.3.1 and things went very well. 

If I go back to Ubuntu and see this again then I will update this ticket. If you prefer to close it, that is fine by me. I can create a new one if I run into this again.

Sorry I cannot provide more info.

Thanks,
Gaston

bubnikv, 2018-08-02:
Thanks for the response. I will close your issue for now.

closed 2018-08-02 by bubnikv