prusa3d/PrusaSlicer #961: [Request]Exclude Out Of Bound Models Automatically

sarf2k4, 2018-06-08:
### Version
1.40.0-beta

### Operating system type + version
Windows 10 64-bit

### Behavior
The current behavior of 1.40.0-beta is that it will automatically disable "export gcode" if there's a model that's not in the printing area. I would like to request that there's an option to "exclude out of bound models automatically" unless it is a required by system especially the prime tower.

Right now, the prime tower is optional unless multi-colour printing were set for the session for a printer that uses multi extruder single nozzle setup. I have mmu 1.0 and always enabled the prime tower but in 1.40.0-beta, for single material/colour mode, the system disabled "export gcode" button if the prime tower are not 100% inside the print area which the prime tower actually would not be generated at all for single mode. 

In summary
1. Enable the export gcode/send to printer button even if the prime tower is not used/generated and sits outside of the print area unless it is generated for multi-extruder/colour printing and sits outside of print area.
2. Exclude models that are out of the print area and do not generate gcode for it and do not include them even in "preview" tab.

What I have in mind actually is about the prime tower as I don't prefer to keep making changes of these profiles. I have 2 printers, mk2 and custom, different bed size and the profile I often used are slightly modified from stock mk2 profiles and can be used with the custom. 

Even If "exclude out of bound models automatically" were to be implemented, I may not able to use slic3r prusa edition to estimate the printed object weight in one go, meaning I have to import one by one and manually calculate the estimate weight of a big print. I have simplify3d, only used for complex and impossible prints but often used for weight estimation of big+many object and it can generate the preview along with the estimation even though it is out of bound.

Maybe you guys can think of a good workaround for this =).

bubnikv, 2018-07-21:
The Slic3r-1.41.0-alpha2 release will ignore objects completely outside the print bed. In that case you will still be able to run the slicing process and the objects outside of the print bed will not be sliced / taken into account for the time / filament weight etc. I believe that covers your request nr. 2.

For the request nr. 1, you are right. We should handle the wipe tower differently from the other objects, either to calculate a minimum wipe tower, or just ignore it. @enricoturri1966, we need to discuss this topic.

Regarding your last request, I believe you can just increase the size of the print bed temporarily. With the new preset UI, the temporary change of the print bed size is clearly indicated and it is easy to return to the old value.

bubnikv, 2018-08-02:
Completely out of bounds objects are excluded from the slicing in Slic3r 1.40.0-alpha3.
Closing.

closed 2018-08-02 by bubnikv