prusa3d/PrusaSlicer #963: Allow config path configuration on linux

Moredread, 2018-06-09:
### Version
1.40.0-beta

### Operating system type + version

 - system: `"x86_64-linux"`
 - host os: `Linux 4.16.14, NixOS, 18.03.132618.0f73fef53a9 (Impala)`
 - multi-user?: `yes`
 - sandbox: `yes`
 - version: `nix-env (Nix) 2.1pre6148_a4aac7f`
 - channels(root): `"nixos-18.03.132618.0f73fef53a9, nixos-unstable-18.09pre140958.696c6bed4e8"`
 - channels(addy): `"nixos-18.03"`
 - nixpkgs: `/nix/var/nix/profiles/per-user/root/channels/nixos/nixpkgs`

### Behavior

On linux the user configuration currently is saved in `~/.Slic3rPE`. 

It would be nice though if it would follow [the XDG Base Directory Specification](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html), esp. the convention for config directories.

This would reduce clutter in the users home directory, and also would allow the user to choose where she saves her config files (via setting the $XDG_CONFIG_HOME).

This is a feature request.

Moredread, 2018-06-09:
See https://github.com/slic3r/Slic3r/issues/832 in the upstream repo, with a possible solution.

vojtechkral, 2018-06-11:
It would be nice if we could follow the XDG recommendation and it would be fairly easy to implement, however, there's the issue of continuity / backwards compatibility. Users already have their configs in `~/.Slic3rPE` and expect them to be there... I'm not sure how we'd go about that.

bubnikv, 2018-06-11:
We can discuss it for the next release, not for 1.40.0

iicurtis, 2019-05-16:
This is now relevant again. Since the rename, we already break backward compatibility.

bubnikv, 2019-05-16:
Are you compiling PrusaSlicer yourself?
Could you try the following?
https://docs.wxwidgets.org/trunk/classwx_standard_paths.html#af5eb9a009b9fd81b748e26d099bb4e89

bubnikv, 2019-05-16:
On the build server to build our binary packages we are using the wxWidgets 3.1.1. This version should support XDG automatically

wxString wxStandardPaths::GetUserDir(Dir userDir) const
{
    // Note that we do not use the file layout here because there is no reason
    // not to respect the XDG convention even if SetFileLayout(FileLayout_XDG)
    // hadn't been called: we're not bound by any backwards compatibility
    // considerations as there can't be any pre-existing config or data files
    // in the home directory that wouldn't be found any longer after updating
    // the version of wxWidgets used by the application.

https://docs.wxwidgets.org/trunk/classwx_standard_paths.html#a86bd7b0187fd299826060f06c7cff5ed

Does it work for you?

iicurtis, 2019-05-16:
I am sorry, I don't quite understand what you want me to test. The current binary still creates a new directory `~/.PrusaSlicer`. I am using wxgtk3 3.0.4 however.

bubnikv, 2019-05-16:
I want you to test the binary slicer (appimage) that we compiled using wxWidgets 3.1.1.

iicurtis, 2019-05-16:
I have tested both the binary in the `.tar.bz2` and the appimage. Both create a directory `~/.PrusaSlicer` instead of using `~/.config`.

guestisp, 2019-05-16:
I can confirm.

```
$ ls -lad ~/.PrusaSlicer
drwxr-xr-x 10 x x 4096 mag 16 13:07 /home/x/.PrusaSlicer
$
$ ls -lad ~/.config/ | grep -i prusa
$ 
```

vojtechkral, 2019-05-16:
We probably need `GetUserConfigDir()` rather than `GetUserDir()` ... I can have a look ...

Schnilz, 2019-10-03:
any news on this ?

jakbyte, 2020-04-21:
Any more news on this?

lukasmatena, 2020-10-26:
This was implemented for PrusaSlicer 2.3.0-alpha2, see #2911. XDG specification should now be respected, users will get an alert when they first run the new version so they know where their config went and where to move it.

Closing.

closed 2020-10-26 by lukasmatena