prusa3d/PrusaSlicer #964: Infill overlap does not seems to have any effect at all in 1.40.0 Beta

fredizzimo, 2018-06-09:
### Version
1.40.0-beta+win64

### Behavior
No matter what I do, it looks like the infill/perimeters overlap settings don't change anything. I don't see any changes neither in the preview nor the actual printed output, even when putting clearly exaggerated values. I also tested with the 1.39.2 version, and I can clearly see the effect there.

There is always some overlap, even with the 0% setting now, and that seems to be bigger than the standard overlap between perimeters for example, as can be seen in the images below.

0% overlap
[image]
100% overlap
[image]

bubnikv, 2018-06-11:
There seems to be a bug with the absolute value. Overlap given by the percent works correctly.

bubnikv, 2018-06-11:
An infill / perimeter overlap was not applied when defined with absolute coordinates. Interestingly nobody used the absolute overlap, as this issue was there for years.

Fixed with https://github.com/prusa3d/Slic3r/commit/6cafc7f121b6ad167d8cc171963f262d986c8704

closed 2018-06-11 by bubnikv