prusa3d/PrusaSlicer #965: Alternatives to overlap/infill gap

fredizzimo, 2018-06-09:
As I see it, there are three reasons for having an infill/overlap setting:
1. To make the perimeters stick better to the walls, especially if there are some positional errors, due to fast movements. I think this is what the parameter is actually meant for.
2. To properly fill solid top layers. No matter which infill you use, there will always be a lot of small gaps, which can usually clearly be seen in the final print. Having a bigger overlap will cause more of those gaps to be filled, without having to tune the extrusion for overextruding the walls. Rather than this, I would like to the see gap fill, fill these in.
3. To fine tune the extrusion multiplier for infill vs perimeters. And usually to have overextrusion at the top layers to create a smoother surface. This is a hack, but it can be used for that. I would rather have more control with separate extrusion multipliers for walls, infill, and top/bottom layers.

So in conculsion my wish is the following:
1. A better gap fill algorithm
2. Separate extrusion multipliers for walls, infill and top/bottom layers. This can partially solve 1., but I it would be much better if the extra filament was adde where it really is needed.

fredizzimo, 2018-06-09:
I just saw the Ironing pull request https://github.com/prusa3d/Slic3r/pull/821 by @supermerill, that seems like it could cover wish 1. and partially wish 2.

stelgenhof, 2018-06-10:
@fredizzimo Are you currently experiencing issues with the overlap setting? Trying combinations of infill pattern and overlap percentage will in most cases result in no gaps.

Please not that the ironing feature is a pure cosmetic one only: it will aid in sticking the infill better to the walls (as per your point number 1).

supermerill, 2018-06-10:
for the wish 3, you can increase the "bridge compensation" parameter in my pull request, it's here to overextrude when you are over a bridged area, it's the main source (imho) of the top surface underextrusion.

You can try it with the link i posted in. I will do a updated one when i will have the time to do so.

supermerill, 2018-06-27:
> 1-To make the **infill** stick better to the walls, especially if there are some positional errors, due to fast movements. I think this is what the parameter is actually meant for.
I agree it's the main reason

> 2-To properly fill solid top layers
I have created an experimental option to tune the flow to have a exact 100% volume extruded for solid  (not-bridge) infill : "enforce full fill volume" on the infill config part. You can find a compiled version here https://github.com/prusa3d/Slic3r/pull/821.
Can you test it? 

3- see above

closed 2026-09-29 by prusaslicer-bot[bot]