prusa3d/PrusaSlicer #967: layer_num placeholder in between object custom gcode

paukstelis, 2018-06-10:
### Version
1.40.0-beta

### Operating system type + version
Linux

### Behavior
* Between object custom gcode does not accept [layer_num] placeholder, resulting in:

> !!!!! Failed to process the custom G-code template between_objects_gcode
> Parsing error at line 1: Variable does not exist
> ; layer [layer_num]

* Verbose gcode shows `; move to next layer (133)` several lines later, and using Before/After Layer custom gcode with `[layer_num]` works correctly.

bubnikv, 2018-06-10:
what layer number do you expect to receive between the objects? What is
your use case?

paukstelis, 2018-06-10:
I would expect to get the last layer before the switch. In the above example it would be 132. 
My use case was looking for a place holder that would change between objects it allow use of [this](https://github.com/paukstelis/Octoprint-Cancelobject) octoprint plugin. Looking through the source I found `[current_object_idx]` which does the trick for getting this to work with sequential printing. What would be even better would be object-specific comment tags included for all layers so it could also work with non-sequential prints :)

bubnikv, 2019-01-29:
> Looking through the source I found [current_object_idx] which does the trick for getting this to work with sequential printing.

Is your issue solved with current_object_idx then?

> What would be even better would be object-specific comment tags included for all layers so it could also work with non-sequential prints :)

Sorry, I don't understand.

paukstelis, 2019-01-30:
This was just an initial work-around attempt for the feature requested in https://github.com/prusa3d/Slic3r/issues/972 and implemented by https://github.com/prusa3d/Slic3r/commit/59ea6431de6da65212d2e068c8cccb4827c978e9

closed 2019-01-30 by paukstelis