prusa3d/PrusaSlicer #968: [Request] Add print time, Material cost and Infill desity editing

LordElijah, 2018-06-10:
Add print time - I would like this to be added to the info box  bellow the right side of the UI. This will show how long the print time of a model will be.

Material Cost - I would like this to be added to the info box below the right side of the UI. This will show the cost of building the model via the price of a spool of filament per kilo.

Infill density editing - This feature is similar to "Layer editing" but instead of editing the layers, the infill density is varied in the different layers of the model.

OBSchofieldUK, 2018-06-10:
As far as I am aware, both print time and material cost are already implemented. 
Print time is estimated when the GCode is generated and exported (under sliced info) 
The cost of the filament can be configured under the filament settings tab - You will need to input the density of the material and the cost per kilo (arbitrary monetary value). Again, this will be displayed once you export Gcode/ send to Octoprin
[image]
t.

tracernz, 2018-06-10:
@OBSchofieldUK covered points 1 & 2.

> Infill density editing - This feature is similar to "Layer editing" but instead of editing the layers, the infill density is varied in the different layers of the model.

Isn't this possible already with modifiers? Here's an old video showing how to do it (disclaimer: I watched with audio muted): https://www.youtube.com/watch?v=FBKgqn4CuHY Now you can create "generic" modifier meshes (cube, cylinder, etc.) from within the edit dialog in Slic3r rather than creating them in external programs (that's still possible too if you need more complex shapes).

scalablecory, 2018-06-11:
Modifiers are the way to go here for infill density, though it is very tedious to use on non-trivial models. An automatic variable density like Cura would rock.

bubnikv, 2018-06-11:
@LordElijah It is good to not mix different proposals in a single issue.

We are working on getting the time and filament consumption shown once the model is sliced without having to export the G-code. There are many issues with the same request, for example this: https://github.com/prusa3d/Slic3r/issues/953

In regard to the adaptive infill, this has been discussed multiple times as well.
https://github.com/prusa3d/Slic3r/issues/290

I am closing this issue, please continue discussing these issues at the existing threads.

closed 2018-06-11 by bubnikv