prusa3d/PrusaSlicer #969: [Request] Dont dsicard all modified settings when switching profiles

Technofrikus, 2018-06-10:
### Version
1.4.0.0beta+

### Operating system type + version
Mac OSX 10.13.4

###Feature Request
When changing profiles after changing some settings, Slic3r asks if you want to discard these changes because the other profile would change them. You can only change the profile and loose all the changed settings or not change the profile at all. **There should be an option to keep these changed settings AND change the profile.** 

It is like that in Cura and saves a lot of time when checking for example how long a print would take with different nozzle sizes and layer heights and not the standard profile settings.

[image]

bubnikv, 2018-06-11:
Maybe I don't quite understand your proposal. Why don't you just clock no, then save the profile and then switch the profile? Is this sequence of steps too long to you?

Technofrikus, 2018-06-11:
This would be a solution, you are right. But for me personally not a good one. There is always one or two settings that I change because of the specific part I want to print and the specific geometry or purpose. So it is not really necessary or practical to have a profile for each possible part/combination.

bubnikv, 2018-06-11:
I think I understand you now.

You want to keep many profiles modified without saving, not just the
current one, right?

Technofrikus, 2018-06-11:
Kind of. I want to have one (or very few per Material) base profile that works well. And then change just some settings because this part needs to be very strong so I increase the perimeters. And then I want to check, how long printing this will take with different layer heights and different nozzle size. But every time I change the nozzle size or layer height, I again have to change the perimeters. So this gets frustrating pretty quickly, especially when there is more then just one setting I changed.

But making a different "strong profile" with more perimeters doesnt make sense for me either, because the next part has a different wall thickness and needs to support even more weight, so I want to have even more perimeters. So I would have a strong and very strong profile? This would get out of hand very fast if I would add a profile for every scenario I come across. Especially with combinations of settings. Or how do you handle these situations?

Basically I want to change nozzle size and layer height without loosing all other changed settings. This would encourage tinkering with these settings more, especially for beginners, to get a feeling how long a print will take with different settings. Also in combination with the feature that the print time is displayed immediately after slicing and not only after exporting the G code. But I know you guys are already working on that :)

Panayiotis-git, 2018-06-11:
A solution would be to not reset the user modified parameters, when the profile is changed. Then these parameters would be marked as changed or not according to the new selected profile.

tracernz, 2018-06-11:
> A solution would be to not reset the user modified parameters, when the profile is changed. Then these parameters would be marked as changed or not according to the new selected profile.

Then you're no longer loading a profile, just a subset of a profile. I can see that leading to unexpected results for most unsuspecting souls.

There are two solutions IMO:
- keep as-is,
- add a third option to the existing dialog, which keeps the fields that were modified in the previous profile.

Panayiotis-git, 2018-06-12:
The second solution fits perfectly my needs.

Technofrikus, 2018-06-12:
There is the possibility of settings that dont make sense together, that is true, but I think that danger is pretty small. If people change these settings, they should know what they are doing. Or learn from their mistakes.

Cura does it like this, which is a pretty good solution in my opinion:

[image]

bubnikv, 2018-06-12:
ok, it makes sense.

rtyr, 2021-01-21:
Already implemented in PrusaSlicer 2.3.0. Closing.

closed 2021-01-21 by rtyr