prusa3d/PrusaSlicer #974: 1.40.0 - No tooltips available

gdombiak, 2018-06-13:
### Version
1.40.0+ final

### Operating system type + version
Mac OS 10.13.5

### Behavior
I was using 1.40.0 alpha & beta versions on Ubuntu and found tooltips very helpful when editing fields. Now that 1.40.0 final has been released, I updated my version on MacOS so I can stop using VirtualBox for testing the alpha/beta versions. However, I noticed that the convenient tooltips available when editing different fields are not available. No idea if they were available on 1.39.* or not.

_Is this a new feature request?_ MAYBE? ❓

DrLex0, 2018-06-16:
This an issue that has existed for a long time. The tooltips don't work for every first page in each of the tabs (i.e. for ‘Layers and perimeters’ in Print Settings, for ‘Filament’ in Filament Settings, and for ‘General’ in Printer Settings). This issue has been in the mainline Slic3r (#2038) and is still open even though it is no longer valid there since they overhauled the UI.

bubnikv, 2018-06-28:
Finally, @YuSanka was able to work around a bug in wxWidgets, and it works! Fixed with 
https://github.com/prusa3d/Slic3r/commit/5f1f7dcbedcdc884b4cd489562b444ad372b89c9
https://github.com/prusa3d/Slic3r/commit/2cbf5b75db3ce89171a9cb0e39f94c72816e3230

It will be part of the 1.40.1-rc release. Closing.

closed 2018-06-28 by bubnikv

DominikPalo, 2018-09-25:
I still have this issue on version 1.41.0+ (macOS). Sometimes (not always as before) all tooltips stop working on the first pages of tabs and only restarting of the slicer helps.