prusa3d/PrusaSlicer #975: Perl post-processing file not recognized in the last version of Slic3r

Gi0cas, 2018-06-13:
### Version
Slic3r 1.40.0+win64

### Operating system type + version
Windows 10

### Behavior
I want to run post-processing script, but after saving of g-code, Slic3r isn't capable to open my Perl file.
This is my perl file :
`#!/usr/bin/perl -i
 
use strict;
use warnings;

while (my $riga = <>) {
    $riga =~ s/M204 S1000/SET_VELOCITY_LIMIT [ACCEL=1000]/g;
    $riga =~ s/M204 S1800/SET_VELOCITY_LIMIT [ACCEL=1800]/g;
	$riga =~ s/M204 S1500/SET_VELOCITY_LIMIT [ACCEL=1500]/g;
print $riga;
}`

The configuration page in Slicer is set like this image :
[image]

This is the error :
[image]

How i can solve ?

Gi0cas, 2018-06-23:
No one can help me? I made some error or this is really a bug?

bubnikv, 2018-06-26:
Slic3r was running slic3r.exe (or slic3r-console.exe) on Windows instead of the perl interpreter. It should be fixed with
https://github.com/prusa3d/Slic3r/commit/645cc65d2b340752f390359ce3ed09a49b463588
and the fix will be part of the 1.40.1 final release.

Gi0cas, 2018-06-26:
Ok, but now how i can solve? i have to wait future fix or i can made some fix manually to solve this issue?

vojtechkral, 2018-06-27:
@Gi0cas The 1.40.1 release, which should solve your problem, will be out in a couple of days.

bubnikv, 2018-06-28:
It will be part of the Slic3r PE 1.40.1-rc release. Closing.

closed 2018-06-28 by bubnikv

Gi0cas, 2018-07-04:
I installed Slic3r 1.40.1-rc2 and now the error with the Perl script is this:
[image]

I checked the file and gaved all the permission (read, write, execute, complete access) to the user of my pc. What i've missed?

bubnikv, 2018-07-04:
Will it work if you put it into your home directory?

Gi0cas, 2018-07-04:
YES !!
I copied the script file in my home directory and worked! 
Thanks a lot

bubnikv, 2018-07-04:
This is interesting. I am not sure why your system did not allow running the script from c:\ 
We tried various things, like running Slic3r under a Guest user, or putting Slic3r into a localized directory (with non-latin1 characters), and everything worked. Anyway, thanks for the report. I will put it into a FAQ.