prusa3d/PrusaSlicer #979: grep: /tmp/.mount_Slic3rwEfrlP/slic3r.desktop: No such file or directory

aboettger, 2018-06-16:
```
$ ./Slic3rPE-1.40.0.linux64-full-201806121015.AppImage             
grep: /tmp/.mount_Slic3rwEfrlP/slic3r.desktop: No such file or directory
```

```
$ stat .//Slic3rPE-1.40.0.linux64-full-201806121015.AppImage
  File: .//Slic3rPE-1.40.0.linux64-full-201806121015.AppImage
  Size: 17635256        Blocks: 34456      IO Block: 4096   regular file
Device: fd01h/64769d    Inode: 20319545    Links: 1
Access: (0775/-rwxrwxr-x)  Uid: ( 1000/   xxxxx)   Gid: ( 1000/   xxxxx)
Access: 2018-06-16 11:45:12.934107465 +0200
Modify: 2018-06-16 11:45:02.377948484 +0200
Change: 2018-06-16 11:45:12.250097594 +0200
```

bubnikv, 2018-06-26:
is it running though?

tracernz, 2018-06-26:
To see what's inside an AppImage you can pass arg `--appimage-mount` (note the trailing & forks the process into background, which I kill at the end).
```
$ ./Slic3rPE-1.40.0.linux64-full-201806121015.AppImage --appimage-mount &
[1] 19185
/tmp/.mount_Slic3rYQLdp4                                                                                                                      
$ ls -l /tmp/.mount_Slic3rYQLdp4
total 38
-rwx--x--x 1 root root  1396 Jun 12 22:36 AppRun
-rwx------ 1 root root 21520 Jun 12 22:36 libunionpreload.so
-rw------- 1 root root   174 Jun 12 22:36 slic3r.desktop
-rw------- 1 root root 16469 Jun 12 22:36 slic3r.png
drwxr-xr-x 6 root root     0 Jun 12 22:36 usr
$ kill -SIGINT 19185
[1]  + interrupt  ./Downloads/Slic3rPE-1.40.0.linux64-full-201806121015.AppImage
```

bubnikv, 2018-07-21:
Would you please tell me, what does not work apart from the following message? Is the application working? What is broken?

grep: /tmp/.mount_Slic3rwEfrlP/slic3r.desktop: No such file or directory

bubnikv, 2019-01-29:
Single report, no response, closing.

closed 2019-01-29 by bubnikv