prusa3d/PrusaSlicer #981: Feature request: acceleration limits relative to velocity

zbrozek, 2018-06-17:
It would be interesting to try setting acceleration relative to the current path. So for example, there may be a higher limit for acceleration in the current direction vs orthogonal to it. I have little idea how much of the work of executing trajectories and handling the actual machine acceleration is done at print time by firmware vs at gcode generation by slicer, so this may be ridiculously infeasible. My goal is to accelerate faster on straights.

Maybe not useful. Maybe super useful. Might be nice to play with.

scalablecory, 2018-06-20:
I'm pretty sure this already happens in firmware.

Sebastianv650, 2018-06-21:
@zbrozek your title suggest something different than that you are describing in your text. Do you mean acceleration depending on movement speed or movement vector?

An acceleration depending on movement speed makes no sense in my opinion. Acceleration depending on movement direction is already checked, as you can set a max. acceleration for each axis (as absolute limits) and separate limits for print and travel moves, this last ones will always apply along the path itself.

zbrozek, 2018-06-21:
Yes; I meant acceleration specifically with respect to components of velocity (meaning the motion vector) as opposed to speed (its magnitude). slic3r has a lot of depth and it's quite likely I'm just overlooking a preexisting feature, but I don't see the capability mentioned [in the manual](http://manual.slic3r.org/expert-mode/speed), nor in the speed settings page within the print settings tab. Maybe it's just somewhere else?

Thanks!

Sebastianv650, 2018-06-22:
That's something which is handled inside the printer FW, not slicer. Slicer can only tell the FW to use a specific acceleration for a move, but if that's also used in reality is depending on the FW limits I mentioned before.