prusa3d/PrusaSlicer #982: Object fits the print volume test should consider also brim

CristianLeonRO, 2018-06-19:
### Version 1.40.0+win64 Slic3r PE

### Operating system type + version
Windows Pro 10 64bit 

### Behavior
* Objects have brim that exceeds the print area and the print volume test introduced in Slic3r PE 1.40.0 should consider also the brim (check attached screenshot or 3MF file 
[LapTop.zip](https://github.com/prusa3d/Slic3r/files/2113850/LapTop.zip)

* _Expected Results_
When brim is enabled, after slicing provide a warning that print is outside of the set volume bounds

* _Actual Results_
Check attached screenshot
[image]

_Is this a new feature request?_
No, a bugfix 

#### STL/Config (.ZIP) where problem occurs
[config.zip](https://github.com/prusa3d/Slic3r/files/2113854/config.zip)

vojtechkral, 2018-06-19:
This might be a dupe of #849

CristianLeonRO, 2018-06-19:
You are right, seems the same issue.

bubnikv, 2018-06-19:
Closing this one, keeping #849 open.

closed 2018-06-19 by bubnikv