prusa3d/PrusaSlicer #984: Supports printed too slowly.

mkbel, 2018-06-19:
### Version
1.40.0 linux64

### Operating system type + version
debian testing amd64

### Behavior
As supports have different layer height, it is considered different layer and minimum layer duration is applied on support separately, even though two layers of support are not printed one after another, but are interlaced with object layers.
* _Steps needed to reproduce the problem_
Object with small support volume per layer.
* _Expected Results_
Slowing down is not applied on support layer separately, if it is followed by object layer. 
* _Actual Results_
Slowing down of print move to honour minimum layer print time, even though next layer to be printed is not support layer again.

_Is this a new feature request?_
It's more like a bug in algorithm.

#### STL/Config (.ZIP) where problem occurs
default config.

top-gun, 2020-09-16:
An addition: The printable object in the same laser height is printed slower than necessary, too. On complex objects, this issue costs a lot of time as smart cooling slows down entire layers down to the configured minimum speed.

When support and Object have the same height, they should be counted together for minimum laxer time.

bubnikv, 2021-02-16:
Fixed with 6ea57817099ca72139bfa3a46fd1dd8a03749734

closed 2021-02-16 by bubnikv