prusa3d/PrusaSlicer #986: Blank Preview and missing part in Layers

habroland, 2018-06-20:
### Version
_Version of Slic3r used goes here_1.40.0+win64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_win10

### Behavior
* _Describe the problem_ Part disappeared from "Preview". The "Layers" tab is completely empty (white)
* _Steps needed to reproduce the problem_ I reduced the layer number to the minimum in the "Preview" tab. Now this tab does not show the part anymore, the scroll bar is gone, and the "Layers" tab is just a white screen. The problem persists even after deleting the whole extracted Slic3r folder AND all the files containing the string "Slic3r" under the AppData folder and subfolders. The issue persist with any stl file.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

[image]
[image]

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-07-23:
@FidelCapo was not able to reproduce this issue. Are there any hints as of how to reproduce this issue? Is this issue specific to a single particular computer? Does it work on other computers correctly?

bubnikv, 2018-12-11:
No response, closing.

closed 2018-12-11 by bubnikv