prusa3d/PrusaSlicer #990: macro language wiki is incomplete

Vladimir-VS, 2018-06-20:
### Version
_Version of Slic3r used goes here_
any
_Use `About->About Slic3r` for release versions_
any
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
any
### Behavior
* _Describe the problem_
the wiki page 
https://github.com/prusa3d/Slic3r/wiki/Slic3r-Prusa-Edition-Macro-Language
is incomplete -- it does not have a reference template for multidimensional vectors (e.g. extruder_offset = 0x0,nx0)
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-06-20:
>    Edition-Macro-Language

Would you please update the wiki?
Thank you.

Vladimir-VS, 2018-06-20:
I'd be more than happy to but I have no clue what is the reference format. I tried all different versions I could think of but all failed. :(

bubnikv, 2018-06-21:
There are two ways of referencing vectors:

The old way
[extruder_offset_0]
[extruder_offset_1]
etc

The new way as part of an expression inside curly braces:
{extruder_offset[0]}
{if extruder_offset[[0] > 12} .... {endif}
etc

Is this clear enough.

Vladimir-VS, 2018-06-21:
Thanks,
Unfortunately, this only addresses the vectors with a single dimension.
Thus, in your example extruder_offset[0] gives a value of, say [17.6,0] and the expression extruder_offset[0] > 12 will produce an error.
The question therefor is -- how to address the extruder_offset placeholder so that it results in a single value of 17.6 in a previous example.

bubnikv, 2018-06-21:
Actually, I don't know from the top of my head. Maybe it is not supported.
I will have to check.

bubnikv, 2018-06-21:
Allright, the bed_shape and extruder_offset are the only variables, which
are vectors of points, and the sad news is, they are not really supported
by the scripting, they are handled as strings. So you may emit the point
into the G-code, but you cannot do arithmetic with the coordinates.

Vladimir-VS, 2018-06-21:
The news is really sad. :(
Is it really hard to implement? Should I move it to requests?

bubnikv, 2018-06-21:
> I have updated the wiki -- please, check if that's ok with you.

that's fine with me.

> Is it really hard to implement? Should I move it to requests?

Let's keep this issue open as a request. It is not really hard, but it takes couple of hours.

Vladimir-VS, 2018-06-21:
ok. thanks. I'll keep the issue open then. As a work-around I'll try to use some irrelevant placeholder as a substitute (e.g. wipe tower parameters).

Vladimir-VS, 2018-06-21:
btw, the link to the list of placeholders is no longer valid :(. Is there any mirror of the one?

bubnikv, 2018-06-21:
https://archive.org/ may help :-)
We should clone it and update it.

Vladimir-VS, 2018-06-21:
thx. posted as a separate page

closed 2026-09-29 by prusaslicer-bot[bot]