prusa3d/PrusaSlicer #992: Feature Request: Z Offset added to Filament Configuration Options

MonkeBear, 2018-06-22:
*New Feature Request*
Add the ability to store Z Offset against a filament profile, as currently it is associated with the printer profile only.

GreatGrizzly, 2018-07-01:
This would be an awesome improvement. Different material types required different amounts of "squish" against the print bed.

PLA requires a closer nozzle then PETG for example.

ardichoke, 2021-09-10:
+1 to this, I'm running into this exact pain point with PETG right now also. It's very annoying to have to adjust my printer profile just to get good adhesion with a different filament type.

An even better improvement would be making it so you can selectively add any setting to the filament override section.

dartrax, 2021-09-20:
Duplicate of #2744

closed 2026-09-29 by prusaslicer-bot[bot]