prusa3d/PrusaSlicer #995: Unable to select a User preset if it's saved as the same name with (modified) appended.

WhiteDragonD, 2018-06-22:
### Version
1.40.0

### Operating system type + version
Windows 7

### Behavior
Saving a user preset with the name:  Prusa PLA (modified)
Now try to select that from the drop down, it will default to the System Preset Prusa PLA instead of the User Modified Preset

- Go into Filament Settings.
- Select Prusa PLA from the System Presets.
- Change the color. 
- Save the new settings name as:  Prusa PLA (modified)
- Now click on the drop down list and try to select the User Preset version of that name.  
- Notice it will move you back to the System Preset option instead of your user preset one.

YuSanka, 2018-06-25:
Actually, you shouldn't use profile name containing substring "(modified)". Because of this one will ruin internaly programm logic. 
But thanks for your report. We added the condition to prevent such cases in the future.

closed 2018-06-25 by YuSanka

bubnikv, 2018-06-25:
Fixed with https://github.com/prusa3d/Slic3r/commit/362a9d8edad4ba02ea476f18428b2dd1a12f5fe6