prusa3d/PrusaSlicer #996: Clicking Preview tab with an empty bed crashes Slic3r

byarmis, 2018-06-22:
### Version
1.40.1-beta

### Operating system type + version
win64

### Behavior
* Open Slic3r
* Click Preview tab in lower left
* Crashes
* Dismiss crash message
* Dismiss crash message again

DrLex0, 2018-06-23:
It's not just with an empty bed, it also often (not always) crashes when going from the Layers tab to Preview after slicing a model. This is on Mac OS X 10.11.6.
Back to 1.40.0 for me…

bubnikv, 2018-06-25:
Our tester @FidelCapo was not able to reproduce the crash. Is there anything special of what you are doing?

@DrLex0 Any chance you can get us a crash log?

DrLex0, 2018-06-25:
I have a reproduction scenario that seems to work reliably:
1. Open Slic3r
2. Load any STL (a small and simple one preferably)
3. Use any print and filament settings
4. Go to the Layers tab
5. Slice it
6. Go to the Preview tab

Then repeat steps 4 to 6 and it goes kablooie.

And I have a crash log as well:
[Crash.zip](https://github.com/prusa3d/Slic3r/files/2132950/Crash.zip)

bubnikv, 2018-06-28:
Fixed in Slic3r PE 1.40.1-rc. Closing.

closed 2018-06-28 by bubnikv