prusa3d/PrusaSlicer #998: "Fill Density", "Support:", and "Brim:" UI text cut-off after an object is added to plater.

Master3See, 2018-06-23:
### Version
1.40.1-beta+linux64

### Operating system type + version
NAME="Ubuntu"
VERSION="18.04 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04 LTS"
VERSION_ID="18.04"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

### Behavior
Summary:
"Fill density:", "Support:", and "Brim:" UI text are cut off after a 3d object is loaded. 

Steps needed to reproduce: 
 - Add an object to the plater to be sliced.

Expected Results:
 -  All UI text should be visible. 

Actual Results: UI text is cut off by 3d preview area. 

[image]

[image]

Master3See, 2018-06-23:
This commit may have caused the issue: [deabeaa](https://github.com/prusa3d/Slic3r/commit/deabeaaa7fc48691c9b778e11a725f7bd93a0458)

closed 2018-06-26 by YuSanka

YuSanka, 2018-06-26:
@Master3See, Thank for your report. We couldn't reproduce your effect of label "cutting", but labels align was incorrect. Please, could you test the fix on your system. Thank in advance )

Master3See, 2018-06-30:
Hi YuSanka,
Confirmed this did fix the issue. Thanks so much!

commit 2fae893a (2018-06-26): Fixed #998

Added detection of gtk2/gtk3 GUI libraries required by Alien::wxWidgets.
Added gtk2/gtk3 include paths, so we may call gtk2/3 API directly if needed for some workaround.