prusa3d/PrusaSlicer #1002: Inconsistent keyboard shortcuts between tabs

DrLex0, 2018-06-24:
### Version
1.40.0+

### Operating system type + version
Mac OS X 10.11.6

### Behavior
In the Layers tab I can change layers with the arrow keys (which is somewhat intuitive). In the Preview tab this functionality is mapped to the ‘d’ and ‘u’ keys. I only discovered this through trial-and-error: I had to press _every_ key on my keyboard to figure this out, due to lack of built-in or online help that explains these keyboard shortcuts. Moreover, I expect ‘browsing’ keys like these to be next to each other. Using the first first letters of ‘down’ and ‘up’ seems to make sense at first, but in video games we don't move our character around with the l, r, f, b keys either.

I would like to have the same keys for browsing through layers in all tabs, and also a built-in help dialog of some kind with a list of all keyboard shortcuts (also see issue #440: the few shortcuts that are supposed to be visible, aren't in Linux and OS X).

_Is this a new feature request?_
Sort of

bobstro, 2018-08-28:
I am having the same issue with Slic3r Prusa Edition 1.41.0-beta+ on macOS High Sierra 10.13.6.

bubnikv, 2018-09-01:
@YuSanka would you please look with @enricoturri1966 into harmonizing the key shortcuts, and possibly to document them on  the Slic3r PE wiki? Thank you.

YuSanka, 2019-01-31:
Completed (full) shortcuts list is implemented now. You can get to know it any time using shortcut "?" or directly from the menu Help ->Keyboard Shortcuts.

closed 2019-01-31 by YuSanka